ROBOTIS-GIT/Dynamixel2Arduino #13: Segmentation fault

sysark-yohan, 2019-07-03:
Hi,

I have a compilation problem with the arduino. I can not understand why I have this problem. All I know is that it comes from the library Dynamixel2Arduino, I looked in the files given by the compiler. But I do not understand where the mistake is.

[image]

sysark-yohan, 2019-07-03:
The error seems to come from the setOperatingMode () function. When I try to switch to OP_CURRENT_BASED_POSITION mode (or any other mode), I get the error above.
On the other hand, I missed something. I use 3 ax-12 and 5 xm430, and sometimes I can not send command to an ax-12 and an xm430, while their ping returns me 1.

OpusK, 2019-07-04:
Hi, @sysark-yohan 

Please tell us which board you are using and the Arduino IDE version.

> I use 3 ax-12 and 5 xm430, 

Are you using only protocol 1.0?

sysark-yohan, 2019-07-04:
I use an arduino mega with the mount in the attachment.

I'm using protocol 1.0 well. I put you the code that I also use as an attachment.

[2019_05_31_FW_CPU_Sysark.zip](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/files/3358159/2019_05_31_FW_CPU_Sysark.zip)
[image]

OpusK, 2019-07-10:
@sysark-yohan 

I'm sorry for late reply.
The compilation error you mentioned did not happen to me.
Is there a problem compiling [the basic example](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/examples/advanced/operating_mode_advanced/operating_mode_advanced.ino)?

>> On the other hand, I missed something. I use 3 ax-12 and 5 xm430, and sometimes I can not send command to an ax-12 and an xm430, while their ping returns me 1.

Are you using broadcast ping?
This may be an issue related to the internal buffer of Arduino. It may be necessary to check if the packet is missing.
Do you have a device like U2D2?
If so, please use [the Packet function of Wizard2.0](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#packet) to check if the packet is exchanged normally.

sysark-yohan, 2019-07-10:
No problem and thank you for taking the time to answer

I do not have the compilation problem anymore. Indeed, I was using the windows application of arduino, I replaced it with the one downloadable on the site of the arduino (.exe), and since I have more probleme.

Actually, I think there is a packet error with the Arduino, so now I use the U2D2 directly with the SDK.

In any case super tool the DYNAMIXEL Wizard 2.0

closed 2019-07-10 by sysark-yohan