ROBOTIS-GIT/Dynamixel2Arduino #21: The library simply does not work

alezarate, 2019-09-25:
Try the following:

XM430-W350-T
12V DC without Jumper (external power)
LED.ino Compilation and upload successful

I changed to AX-12A, changed to protocol 1.0 and nothing.

instead using
https://github.com/descampsa/ardyno
AX-12A with protocol 1.0 the shield works!

what is the difference between the DynamixelShield and Dynamixel2Arduino libraries, is very confusing.

Something am I doing wrong?

OpusK, 2019-09-26:
Hi, @alezarate 

> what is the difference between the DynamixelShield and Dynamixel2Arduino libraries, is very confusing.

As [e-manual](http://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#library-api) explains, only the Shield library uses the Dynamixel2Arduino library. (Inheritance relationship)

> XM430-W350-T
12V DC without Jumper (external power)
LED.ino Compilation and upload successful

> I changed to AX-12A, changed to protocol 1.0 and nothing.

Does the XM430 work fine, but is the AX-12A not working?

alezarate, 2019-09-27:
Hi @OpusK ,

It has not worked with any combination:

OpenCR + XM430-W350-T
Arduino UNO + XM430-W350-T
Arduino MEGA + XM430-W350-T
Arduino UNO + AX-12A
Arduino MEGA + AX-12A
OpenCR + AX-12A

But there are changes in the DATA pin, they can be seen with the oscilloscope.

I didn't make any changes to Baudrate, I use ID: 1 and I also tried with ID: 254.

alezarate, 2019-09-27:
Please see the attached captures, look at the time scales.

One of the captures is using Dynamixel2Arduino (no works) and the other is using ardyno (works).

The graph corresponds to the command to turn on the LED using protocol 1.0 on an AX-12A.

[image]

[image]

alezarate, 2019-09-27:
I apologize.

The new dynamixel (at least the ones bought by the company) have a baudrate of 10000000 factory configured.

The library works fine, thanks for the support.

closed 2019-09-27 by OpusK