ROBOTIS-GIT/Dynamixel2Arduino #22: Slow communication rate for protocol 1.0

chen-tianjian, 2019-09-27:
Hi OpusK,
Thanks for working on this library. It works perfectly for my XL430 and XM430. But for AX12A and AX18A, the control frequency can only reach to 10Hz. (Interestingly, it is precisely 10Hz). I guess it might be a common problem for protocol 1.0, but got lost in reading the code.

Here is my simple testing code:
```
#include <Dynamixel2Arduino.h>
#define DXL_SERIAL   Serial1
#define DEBUG_SERIAL Serial
const uint8_t DXL_DIR_PIN = 2;
const uint8_t DXL_ID = 1;
const float DXL_PROTOCOL_VERSION = 1.0;
Dynamixel2Arduino dxl(DXL_SERIAL, DXL_DIR_PIN);

void setup() {
    DEBUG_SERIAL.begin(115200);
    dxl.begin(1000000);
    dxl.setPortProtocolVersion(DXL_PROTOCOL_VERSION);
    dxl.ping(DXL_ID);
    dxl.torqueOff(DXL_ID);
    dxl.setOperatingMode(DXL_ID, OP_POSITION);
    dxl.torqueOn(DXL_ID);
}

unsigned long prev_time;
void loop() {
    dxl.setGoalPosition(DXL_ID, 512);
    DEBUG_SERIAL.println(micros() - prev_time);
    prev_time = micros();
}
```
The number printed (the time spent in every loop) is around 100000 us = 100 ms. I guess that is too much time for any smooth operation. Do you have an idea why this happens? Or, could you please point out where in the library I should have a look?

(BTW, I am using Teensy 3.6 and a custom shield with pin 2 as the direction pin.)

Best,
Tianjian

OpusK, 2019-09-30:
Hi, @chen-tianjian 

> AX12A and AX18A, the control frequency can only reach to 10Hz. (Interestingly, it is precisely 10Hz). 

It seems to me that the problem is caused by a lack of normal communication with the AX12A.
Internally, the default timeout is set to 100ms, which seems to be an exact match, indicating that a TIMEOUT has occurred.

Use the functions below to check for any errors.
- [getLastStatusPacketError()](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/src/utility/master.h#L174)
- [getLastLibErrCode()](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/src/utility/master.h#L175)

See below for usage examples.
- [advanced/slave.ino](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/examples/advanced/slave/slave.ino#L77)
- [e-manual API manual](http://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#master-class)

chen-tianjian, 2019-10-01:
Thanks @OpusK  for your instructions. I checked the error code and it is TIMEOUT. I guess communication fails in most cases and only succeeds occasionally. I still do not know the reason, but as I factory-reset the motor, it works fine. 
Thank you!

closed 2019-10-01 by chen-tianjian