ROBOTIS-GIT/Dynamixel2Arduino #23: Wishlist: As I am converting from using Dynamixel SDK...

KurtE, 2019-10-13:
Hi @OpusK,

I am trying to get myself going again on the updating of some of Hexapod code base, to now use this library instead of Dynamixel SDK.  The main board that I believe Trossen Robotics will use, is the OpenCM9.04... But I would also for my own usage like for it to work well with Teensy boards by PJRC (with my own carrier boards).   As I do some of this conversions, I run into different things and or wish things were ...  So not sure if this is best place to ask/mention these.   Would you prefer that each by separate, so can easily close off as answered?  

So far some of the things I have seen include:

a) Wish more (all) of the examples had more complete options for different boards.  Example the scan_dynamixel sketch does not have any example of how to configure up to run on OpenCM or OpenCR boards.  I did see OpenCM stuff in read_write_ControlTableItem.ino.

b) Wish you would automatically handle somethings like:

` #define DXL_SERIAL   Serial3 //OpenCM9.04 EXP Board's DXL port Serial. (To use the DXL port on the OpenCM 9.04 board, you must use Serial1 for Serial. And because of the OpenCM 9.04 driver code, you must call Serial1.setDxlMode(true); before dxl.begin();.)`

That is maybe on first call to dxl.begin() if compiled for OpenCM and Serial port is Serial1, do the setDXLMode(True)... Or could test again pin 28 in this case...

c) Wishing for some of the member functions including some of the constructors, you allowed pass by pointer as well as by reference.  Including ones like:
1) Master::setPort  - I may have a set of ports in an arry so may want to do something like: dxl.setPort( with a pointer to port object)... WIll see how hard to get arround.

2) SerialPortHandler::SerialPorthandler - WIsh I could pass in HardwareSerial port by pointer...)

3) Dynamixel2Arduino::Dynamixel2Arduino - maybe should allow to pass in port by address and default to nullptr?   If I going to setup more than one HardwareSerial objects to pass in a different times, then maybe should be able to initially just be able to declare the main object like:
Dynamixel2Arduino dxl;

And when I am ready to use it do something like: 
`dxl.setPort(port_list[index_port_list]);`

That is all for now...

OpusK, 2019-10-14:
Hi, @KurtE 
Thank you for your detailed feedback about this library.

> a) Wish more (all) of the examples had more complete options for different boards. Example the scan_dynamixel sketch does not have any example of how to configure up to run on OpenCM or OpenCR [...]

The examples are pin definitions only on the premise of using DynamixelShield.
Previously, there were contents of OpenCM904, but it was removed due to the mixing of Shield and the general board like OpenCM904.
However, as you say, for your convenience, I'll consider adding a README or an example, providing examples of configuration based on the characteristics of the boards.

> c) Wishing for some of the member functions including some of the constructors, you allowed pass by pointer as well as by reference. Including ones like:

I'll add one more API that takes a pointer parameter. Thank you for your good feedback.

> Dynamixel2Arduino::Dynamixel2Arduino - maybe should allow to pass in port by address and default to nullptr? If I going to setup more than one HardwareSerial objects to pass in a different times, then maybe should be able to initially just be able to declare the main object like:
Dynamixel2Arduino dxl;

The initial design of the APIs was designed only considering the basic circuit of Dynamixel.
Again, I'll add more constructors.

> Wish there was easy way to setup to do other forms of SerialPortHandler (subclass?) That is the current version assumes you are using RX, TX and direction pin... But for example with Teensy, I may wish to either use direction pin, in that case I can tell the SerialPort to handle it for me, or I may wish to use half duplex mode (TX Pin), in which case I need to setup in init and set a register bit for which direction I am going ...

In the case of the Master class, just like the [SerialPortHandler class](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/src/utility/port_handler.h#L43), you can inherit from the PortHandler class to create a new class.
However, the Dynamixel2Arduino class declares and uses SerialPortHandler as a member variable for design purposes.
I'll change it to a Pointer so that the user can create and set up a new class that inherits from SerialPortHandler.

> d) keywords.txt needs a pass through to update with all of the member names like:
setPortProtocolVersionUsingIndex

I'll add this too.

When I update(PR) the above, I will add you as a reviewer.

OpusK, 2019-10-14:
@KurtE 

I have created and updated a branch for some of the issues above.

There are still points to consider or think about more. I added [an example with respect to the custom SerialPortHandler](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/update_for_%2323/examples/advanced/add_custom_SerialPortHandler/add_custom_SerialPortHandler.ino). I hope this example helps you.

KurtE, 2019-10-14:
Hi @OpusK, 

Will take a look at the new branch:

On examples: My understanding from before is that for the Dynamixel Shield, there is already a different library which is a subclass of this library (DynamixelShield), so obviously those examples should be centered around the shield.  However if this library is supposed to be a better light weight library to use for embedded stuff using the OpenCM... then personally think the examples should include this board.  (But that is just my $.02 worth)...

A couple of other things I have or will run into. 

1) I have an OpenCM9.04 controlling a Hexapod robot, which for me the majority of the time the legs have 3dof (on some earlier non Dynamixel Hexapod I had  4dof).  Plus sometimes have a turret with 1 or 2 servos.   So I typically need the ability for about 20 servos.  Currently I believe the defaults for OpenCM is 16.  So need to define some way to expand this.  Yes I could edit your library and change your defaults, but this implies that everyone who uses this code will have to do the same. 

2) How to use the more extended versions of ping.  That is: 
```
XelInfoFromPing_t ping_info;
if (dxl.ping(i, &ping_info, 1)) {
```
Will fail to compile, likewise:

```
RecvInfoFromPing_t ping_info_list;
dxl.ping(DXL_BROADCAST_ID, ping_info_list);
```

Something about C++ when a derived class has a member function with same name as base class, the base class ones are not usable...  

I have hacked up a version that compiles, (but I have not tried running yet. 
`if (((DYNAMIXEL::Master*)&dxl)->ping(i, &ping_info, 1)) {`
But hope there is a better way?

KurtE, 2019-10-14:
@OpusK, 

Forgot to ask, should I do more comments here about changes and/or part of the PR? 

Memory(OpenCM) - I understand memory is limited (20K? ) So obviously can not go totally wild on using memory. 

So again, the question comes up with what is a valid default size for DXL_MAX_NODE and what does changing that impact and does it make sense to maybe change this separate some of these limitations? 
a) Probably most important:  The map of ID to Model Numbers in Dynamixel2Arduino.h:
    IdAndModelNum_t registered_dxl_[DXL_MAX_NODE];

Currently for openCM board I believe this now takes: 4 bytes per item (as unpacked) so 64 bytes, and if you for example increased this to 32 instead of 16 128 bytes...  Could again reduce this by allowing the structure to be packed, example using pragma packed.  

Or could go whole hog and allow all 256(or 253) ids and make it a simple array of Model numbers indexed by ID...   

And/Or could instead of storing whole ModelNumbers could convert these maybe to a 1 byte value of something like ModelIndex  so you don't have to do large case statements for each usage.

b) Ping: again above you could have a default MAX number allowed to be pinged, which again assuming you call the XEL version forms of ping, then you gave me an out, of simply defining my own array to pass in, which solves that... In fact personally I don't see the need to have one with the hard coded max size... 

c) SyncWrite (and likewise syncRead probably bulk ones as well).  But the main one I currently use is SYncWrite as it was there for Protocol 1... 

Personally I don't like the current APIS.  Example suppose again suppose I edit DXL_MAX_NODE to just handle the default hexapod with lets say 20 servos. 

In order to use the SyncWrite to output the new positions for the servos.  
The SyncWrite code is setup that I have: 
```
  typedef struct XelInfoForSyncWriteParam{
    uint8_t id;
    uint8_t data[DXL_MAX_NODE_BUFFER_SIZE];
  } XelInfoForSyncWriteParam_t;

```
And currently DXL_MAX_NODE_BUFFER_SIZE is 16.
So ParamForSyncWriteInst_t sync_write_param;
Will take up something like: 20*17+9 bytes: 349 bytes. 

And in order for you to actually send out the data, you have to then pack the data up and you reserved a buffer in the DxlPacket to handle this (sort of), and you have two of those structures as part of dxl as well. 
Note: DXL_BUF_LENGTH would be:  20*16+11= 331
#define DXL_BUF_LENGTH (DXL_MAX_NODE*DXL_MAX_NODE_BUFFER_SIZE + 11

So will be looking to see if I can do the packing myself, and avoiding double packing. 
Sort of along the line of : begin sync write, set sync write data (index, buffer), send it now...

Side note: currently if by the real off chance that someone tries to send 16 bytes of data to all 16 max servos, your packet buffer won't be large enough as you need an extra byte per servo for the id.

Now back to experimenting

KurtE, 2019-10-14:
@OpusK  - On the ping issue above, 

I think the fix for it is to add a using statement in Dynamixel2Arduino.h 
That is: 
```
 using DYNAMIXEL::Master::ping;
    bool ping(uint8_t id = DXL_BROADCAST_ID);
```

I put it just above the new ping member here, but probably could be in any location within the public: 
area

OpusK, 2019-10-15:
@KurtE 

Thanks for your feedback :)
First of all, I'll explain to you the design concept related to APIs.

1) `Dynamixel2Arduino` class is for Arduino novice users. Therefore, even if memory waste occurs, users do not have to worry about memory management. However, as you said, I think it's a very inconvenient API for users who can manage memory.

2) The `Master` class has a more raw API than the Dynamixel2Arduino class. In order to solve some of the memory-related absurdities of `Dynamixel2Arduino`, I created a form that inherits. 

Anyway, an important concept is that Dynamixel2Arduino's API should be easy to use for Arduino novice users. 

I and our team had a lot of trouble with this, but now we're just designing the API to this level.
In fact, I still think there's a way to change APIs and memory structures more easily and efficiently.
Therefore, I still leave the major version at 0 (0.x.x). ;)
So, I welcome and appreciate your feedback.

With the feedback you gave me, I will think about it and leave an additional reply.

OpusK, 2019-10-15:
@KurtE 

The following has been updated so far.
> On examples: My understanding from before is that for the Dynamixel Shield, there is already a different library which is a subclass of this library (DynamixelShield), so obviously those examples [...]

- [Associated commit](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/pull/24/commits/90ad884846c2284bf59555f87eba7ad00f5fa153)

> On the ping issue above,

- [Associated commit](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/pull/24/commits/cab63edd8d394c87d48ff9a8215d645ee20233c1)

> several methods in Dynamixel2Arduino.h comments mention getLastErrorCode() which I don't think exists. But instead in master.h is: getLastLibErrCode

- [Associated commit](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/pull/24/commits/cab63edd8d394c87d48ff9a8215d645ee20233c1)

In addition, the contents related to memory management (Model Index, etc.) need more discussion and consideration.

KurtE, 2019-10-15:
@OpusK, 

Here is a modified version of the sync_read_write_raw or some such name, that I have two other files included in sketch that so far defines a possible sync write object...   It won't run yet as I am not sure how to call something like: 
`    error_code = dxlTxPacketInst(&packet_, DXL_BROADCAST_ID, INST_SYNC_WRITE, _buffer,  _cnt_servos * (_node_size+1) + 4);
`
But you might take a look at the class and changes to sketch to see if this makes sense and would be a easy way to go forward...
[Dynamixel2Arduino_sync_read_write-191015a.zip](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/files/3730859/Dynamixel2Arduino_sync_read_write-191015a.zip)

KurtE, 2019-10-15:
Quick update: I hacked up your current version of library (master.h and Master.cpp) and added:
// Experiment allowing some of the TX, RX functions to be called
```
 bool  Master::txPacketInst(uint8_t id, uint8_t inst_cmd, uint8_t *p_data, uint16_t length) {
  uint32_t pre_time_us = micros();
  last_lib_err_code_ = dxlTxPacketInst(&packet_, id, inst_cmd, p_data, length);
  packet_.tx_time = micros() - pre_time_us;
  return last_lib_err_code_ == DXL_LIB_OK;
 }

And edited the send method of the class I defined in the above zip file:
```
bool  SyncWrite::send(Dynamixel2Arduino &dxl) {
  // We need to fill in the header information for ths sync write.
  if (!_buffer) return false;

  //uint32_t pre_time_us = micros();
  //lib_err_code_t error_code;

  if (dxl.getPortProtocolVersion() == 1.0) {
    //protocol 1
    _buffer[2] = _starting_addr & 0xff;
    _buffer[3] = _node_size & 0xff;

    return dxl.txPacketInst(DXL_BROADCAST_ID, DYNAMIXEL::INST_SYNC_WRITE, &_buffer[2], _cnt_servos * (_node_size + 1) + 2);
  }
  //protocol 2
  _buffer[0] = _starting_addr & 0xff;
  _buffer[1] = _starting_addr >> 8;
  _buffer[2] = _node_size & 0xff;
  _buffer[3] = _node_size >> 8;
  return dxl.txPacketInst(DXL_BROADCAST_ID, DYNAMIXEL::INST_SYNC_WRITE, _buffer,  _cnt_servos * (_node_size + 1) + 4);
}
```
It builds, but I have not yet tried it... 

```

KurtE, 2019-10-17:
Hi @OpusK,

I had a chance to play around some more with the Sync Read/Write code I mentioned yesterday and made them into classes.  These classes allow you to have them allocate memory for them by default or you can pre allocate memory and pass in the buffer to use.

I have update the a version of the sync read/write sketch to do simply set servo goal positions and query back current positions. 
[Dynamixel2Arduino_sync_read_write-191016a.zip](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/files/3737087/Dynamixel2Arduino_sync_read_write-191016a.zip)

The above sketch is also in my github project I mentioned earlier:  https://github.com/KurtE/Open_CM_CR_Arduino_Sketches/tree/Dynamixel2Arduino

It required a few changes to master.h and master.cpp to allow external libraries/code to be able to call functions to send and receive packets:
master.h I added:
```
    // Experiment allow external code to call tx, rx functions
   bool  txPacketInst(uint8_t id, uint8_t inst_cmd, uint8_t *p_data, uint16_t length);
   const dxl_packet_t *rxPacket();

```

And in Master.cpp: 
```
// Experiment allowing some of the TX, RX functions to be called
 bool  Master::txPacketInst(uint8_t id, uint8_t inst_cmd, uint8_t *p_data, uint16_t length) {
  uint32_t pre_time_us = micros();
  last_lib_err_code_ = dxlTxPacketInst(&packet_, id, inst_cmd, p_data, length);
  packet_.tx_time = micros() - pre_time_us;
  return last_lib_err_code_ == DXL_LIB_OK;
 }

const dxl_packet_t *Master::rxPacket() {
  uint32_t pre_time_us = micros();
  last_lib_err_code_ = dxlRxPacket(&packet_);

  // See if we received some valid data...
  if(last_lib_err_code_ != DXL_LIB_OK) 
    return nullptr;
  packet_.rx_time = micros() - pre_time_us;
  return &packet_.rx;
}

```

Currently the sketch is moving I think 12 servos to 2048+-256.

If you get a chance you might take a look and see what you think.

Also still personally I think it would be great if you hid some of the special case code for  OpenCM board to make simple code for Seria1 work without needing user code to do something special.
Example you could do it something like:

```
void SerialPortHandler::begin(unsigned long baud)
{
#if defined(BOARD_OpenCM904)
  if (_dir_pin == 28) {
      Serial1.setDxlMode(true);
  }
#endif

  baud_ = baud;
  port_.begin(baud_);
  
  if(dir_pin_ != -1){
    pinMode(dir_pin_, OUTPUT);
    digitalWrite(dir_pin_, LOW);
    while(digitalRead(dir_pin_) != LOW);
  }
}
```
With the code in the #if defined added...

OpusK, 2019-10-17:
Hi, @KurtE 

Thanks for your feedback and contribution.
I have considered your various proposals and the problems you presented earlier.. (with multi-platform support, etc..)

First of all, Master Class should be modified to change the APIs of Sync and Bulk.

Therefore, if we need to modify the API anyway, I would like to modify the API a lot for the following purposes:
  - Support multi-platform (Arduino, bare-metal, OS, etc)
    - I want to write functions such as Master Class in c language for portability.
      - Of course, for Master, I'll wrap it in CPP (class) (with a similar API now).
  - Reasonable memory (buffer) allocation
    - As you mentioned, the current memory structure can be stable, but it's inefficient. So, as you suggested, it will support dynamic allocation and custom buffer selection.

Of course, the functions of the Dynamixel2Arduino class will not change. However, if the Master is partially changed and the Sync and Bulk classes are created as you suggested, the related advanced examples will inevitably be modified.

If you have any ideas or opinions as you have been, please tell me.
Thank you again for your contribution.

OpusK, 2019-10-17:
@KurtE 

https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/commit/d49c0fe30bf6ab00075388206afef2653e9bb197

Before changing the layer, I merged some of the suggestions you mentioned in the develop branch.
 - OpenCM9.04 setDxlMode in begin of SerialPortHandler.
 - raw APIs of Master
 - packing of IdAndModelNum structure.

KurtE, 2019-10-17:
@OpusK 

Thanks for merging all of this stuff into the library.   Wonder if we should continue on this issue or startup a new one...

Personally I don't see a need to convert to C, especially with things like PortHandler... As if you wish to support multiple handlers with function pointers, you are now just emulating c++ without the compile type checking.   I understand there are some C programmers out there that may wish to do DXL programming.  For example: over the last year or so, with the Teensy stuff for the new T4, I sort of reversed some of these versus the T3.x branch.  Example HardwareSerial class with T3.x the code was developed like for Serial1, there may have been a c function  like: int serial1_available(), which would return the count and the C++ class was simply  HardwareSerial1::available() {return serial1_available();}

Now with T4 code we have HardwareSerial::available() doing the work, and a Serial wrappers file (in c++) do something like: 
```
extern "c" {
    int serial1_available() {return Serial1.available();}
}
```
Again side note: But on T4 code base did some additional things to minimize memory usage with this.
in particular there is HardwareSerial.h (and .cpp), but in addition there is HardwareSerial1.cpp, HardwareSerial2.cpp... HardwareSerial7.cpp  

With this for example: HardwareSerial1.cpp is setup  to define Serial1 object, plus all of the default buffers it uses.  I also put hooks into the main code that before might look something like: 
```
void main () {
    setup();
    for(;;) {
        loop();
        if (Serial.available()) serialEvent();
        if (Serial1.available()) serial1Event();
        if (Serial2.available()) serial2Event();
...
        if (Serial7.available()) serial7Event(); 
    }
}
```
That instead of explicitly calling the SerialX.available()... functions I instead call on static member of HardwareSerial, where I keep a list of which SerialX.begin() methods have been called.  The begins set a function pointer to call as part of their begin.  Note: their weak linked function like Serial1Event if called, sets the function pointer back to NULL to not call again...  

Why this and why mentioning here?  By doing the above, that when you build a sketch that for example does not user Serial2-Serial7,  these objects and any necessary buffers are not allocated in that sketch, nor is the overhead then of calling their event code.  Although the main reason for indirect calling of the SerialX.available... was to not have any explicit reference to the object which would always bring everything in... 

So if you are reworking some of the things like the Master class, you may want to look at what things are tightly coupled into the object, especially if they bring in any additional things like buffers...  Example I have never used the Bulk Read and Write stuff.

---

More back to this library and: Support multi-platform (Arduino, bare-metal, OS, etc) 

What may turn into an interest challenge is things like which of these platforms support big endian versus little endian?    This is another reason why I was migrating over to using methods (functions) to move data in and out of SyncRead/SyncWrite.  Example code like this out of sync_read_write_raw
`memcpy(sync_write_param.xel[0].data, &velocity, sizeof(velocity));`
Will not work properly on the other platforms. 

I have not fixed that yet in my class I posted, where instead I do something like:
`syncwrite.setItemByIndex(i, &position);`

Note: My method also has optional parameters for offset within the data to write as well as size to copy.  So if I wish to handle this properly I could add one or more ways to do this.  Could do maybe something like explicitly say it is a 4 byte field: 
`syncwrite.setItemByIndex4(i, &position);`
Or could have method where I explicitly pass in the starting register number like: 
`syncwrite.setItemByIndexRegister(i, &position, offset,  DXL_X_GOAL_POSITION);`
With this could make passing in the register number optional, as the constructor gives me the starting register number... Of course you may have to do it explicit if using indirect registers.... 

---

I still wish/want/need support for > 16 servos on OpenCM hopefully by default. 

Personally I recommend maybe slightly decoupling of MAX_NODE and MAX_NODE_BUFFER_SIZE from DXL_BUF_LENGTH  and/or turn things upside down a little... 

That is in config.h where you have something like:
```
#if defined (ARDUINO_AVR_UNO) || defined (ARDUINO_AVR_YUN) \
  || defined (ARDUINO_AVR_INDUSTRIAL101)
#define DXL_MAX_NODE                   16
#define DXL_MAX_NODE_BUFFER_SIZE       10
#elif defined (ARDUINO_AVR_LEONARDO)
#define DXL_MAX_NODE                   16
#define DXL_MAX_NODE_BUFFER_SIZE       12
#elif defined (OpenCR)
#define DXL_MAX_NODE                  253 // Max number of XEL on DYNAMIXEL protocol
#define DXL_MAX_NODE_BUFFER_SIZE       32
[... 9 lines cut ...]
Wish maybe it was more like:
```
#if defined (ARDUINO_AVR_UNO) || defined (ARDUINO_AVR_YUN) \
  || defined (ARDUINO_AVR_INDUSTRIAL101)
#define DXL_MAX_NODE                   16
#define DXL_BUF_LENGTH                  171 
#elif defined (ARDUINO_AVR_LEONARDO)
#define DXL_MAX_NODE                   16
#define DXL_BUF_LENGTH                  203
#elif defined (OpenCR)
#define DXL_MAX_NODE                  253 // Max number of XEL on DYNAMIXEL protocol
#define DXL_BUF_LENGTH                  8107
#elif defined(Arduino_OpenCM904)
#define DXL_MAX_NODE                   24
#define DXL_BUF_LENGTH                  256    // actually could be 251 to give 10....
#else
#define DXL_MAX_NODE                   16
#define DXL_BUF_LENGTH                  267
#endif

#define DXL_MAX_NODE_BUFFER_SIZE     ((DXL_BUF_LENGTH-11)/ DXL_MAX_NODE)
// 11 = Header(3)+Reserved(1)+ID(1)+Length(2)+Instruction(1)+Error(1)+crc(2)

```
Probably enough rambling for this round...

OpusK, 2019-10-18:
@KurtE 

> Wonder if we should continue on this issue or startup a new one...

I think it's a good idea to discuss the task of converting to c, memory issues, multiplatform support, etc. in a new thread.

> Forgot to mention, it looks like there is no implementation for:
C:\Users\kurte\AppData\Local\Temp\arduino_build_578653\sketch/syncReadWrite.cpp:226: warning: undefined reference to DYNAMIXEL::Master::setLastLibErrCode(DYNAMIXEL::LibErrorCode)'`

Oh, my mistake. I'll fix it.

> #if defined (ARDUINO_AVR_UNO) || defined (ARDUINO_AVR_YUN) \
  || defined (ARDUINO_AVR_INDUSTRIAL101)
#define DXL_MAX_NODE                   16
#define DXL_BUF_LENGTH                  171 
#elif defined (ARDUINO_AVR_LEONARDO)
#define DXL_MAX_NODE                   16
#define DXL_BUF_LENGTH                  203

I understand your opinion, but can you tell me why each buffer length is the same as above? I understand 16 and 171 (16 * 10 + 11), but I want to know what standard you did this way. (That is, a comment is required)

> Personally I don't see a need to convert to C

Apart from the Arduino, we want to convert the code to c in case the user wants to use c only (c compiler only) on other platforms.

KurtE, 2019-10-18:
@OpusK 
New thread it shall be...

As for the changes of defines like I mentioned.  Some of it is just my gut reaction about what things are driving what things...   That is currently the DXL_BUF_LENGTH is defined to in theory to handle the maximum operation like Sync Write or Sync Read.  And it is assumed that all the servos that you may have will be part of that sync write... 

Again as I think I sort of mentioned earlier is also not 100% correct. 

That is for Sync_Write, your data structures are:
```
typedef struct XelInfoForSyncWriteParam{
  uint8_t id;
  uint8_t data[DXL_MAX_NODE_BUFFER_SIZE];
} XelInfoForSyncWriteParam_t;

typedef struct ParamForSyncWriteInst{
  uint16_t addr;
  uint16_t length;
  uint8_t id_count;
  XelInfoForSyncWriteParam_t xel[DXL_MAX_NODE];
} ParamForSyncWriteInst_t;
```
The current defined buffer size is:
`#define DXL_BUF_LENGTH (DXL_MAX_NODE*DXL_MAX_NODE_BUFFER_SIZE + 11) // 11 = Header(3)+Reserved(1)+ID(1)+Length(2)+Instruction(1)+Error(1)+crc(2)
`
So your actual maximum packet would be: Header(3)+Reserved(1)+ID(1)+Length(2)+Instruction(1)+StartingAddress(2)+datalength(2)+DXL_MAX_NODE*(ID(1) + DXL_MAX_NODE_BUFFER_SIZE)+crc(2)
Or for example 16 ids with max 16 data per servo.
14 + 16*17 = 286

Where current actual define is 267

And it is even worse for bulk write as you have ID, addr, length plus data for each node...

So again my comments were just more of a gut reaction.    Would more likely  change definitions to maybe have MAX IDs separate than number of ids in Sync or bulk operations...  As there may be lots of apps who play with servo who never use Sync or Bulk.... 

But again just rambling

Again I was trying to semi decouple the BUFFER sizes from the off chance you are going to do a

KurtE, 2019-10-19:
@OpusK,

Maybe should be on new Issue or Forum or email or ??? 

If you are interested, for the fun of it, with the sketch I had with potential C++ Sync read and write code, I wrote another version of Sync Read in C.  There are two new files that are part of that sketch, and the INO file calls these as well. 

Again this is only a test looks like it may be sort of working, obviously if it were real for library would need cleanup and ... 

It is up at: https://github.com/KurtE/Open_CM_CR_Arduino_Sketches/tree/Dynamixel2Arduino/Dynamixel2Arduino_sync_read_write

I will probably won't do anything else over the weekend or so... 

But the nice thing is, I can do all of this external from library.  So only issue left in library is for the default number of IDs that it can map ID to model number... 

Hope you have a good weekend

OpusK, 2019-10-21:
@KurtE 

> Maybe should be on new Issue or Forum or email or ???

I think it would be nice to proceed on GitHub issue.

> If you are interested, for the fun of it, with the sketch I had with potential C++ Sync read and write code, I wrote another version of Sync Read in C. There are two new files that are part of that sketch, and the INO file calls these as well.

Thanks for sharing!
Currently, I am reworking the functions in protocol.cpp in C style. (Including buffer related)
Master, slave, portHandler, etc. will try to reconsider afterward.
Sooner or later I want to put this on a new branch.
I'll review the C version of your syncRead later, as we review it together.

KurtE, 2019-10-24:
@OpusK - I am back from a mini vacation and now recovering ;) 

For now I will probably not do much more with the Sync Read/Write, other than keep it in the back of my mind that it is easily doable to not need to use up so much memory on these operations.

So I might now look at how to increase the number of servos supported without killing memory. 

Then I may hack up a version of SerialPort - that supports other boards, with or without using Hardware Buffer chips.   

My exercise in converting the C++ SyncRead into a C version reminded me several reasons why I don't do straight C anymore.  Note: I am not one to use C++ bells ans whistles like the STD libraries, or templates, or dynamic casting, or ... 

But some things I do find come in handy:
a) Class and Instance variables and to some extent name space:   Old days would create a structure or structures with the member variables and used lots of global variables that could easily conflict...  Likewise for conflicts of function names.   Then remembering to have to pass in a pointer to the data structure to all the functions associated with the "class"

VTables - In the old days I did do things like, your data structure would contain either a set of member  variables or pointer to table of pointers, that you would call.  So If you wanted a sub class, you new you had to update the 4th pointer to point to your method...  Again all you are doing is emulating C++ here. 

Optional/default Arguments on functions:  That is these days, I might do something like:
` void begin(uint32_t speed=1000000, HardwareSerial *pserial=Serial1, int direction_pin=-1);`
And in C++ you would maybe have it setup where you could most of the time get away with simply calling begin () with no parameters...

But in C there is no such thing as default arguments.

Overloading of functions, in C++ I can have two different call signatures by the same name.   A simple case is you go well maybe I could do the begin above, by making a couple of versions of it:
```
void begin();
void begin(uint32_t speed);
void begin(uint32_t speed, HardwareSerial *pserial, int direction_pin);
```
But that again is not allowed. 

So I am keeping my fingers crossed that you won't completely sacrifice the C++ stuff for C.  

I am still sort of curious on what platform/setup that people are requiring only C?  I know that on many platforms they use GCC compilers, so should also have both compilers...   But I understand, you got to do, what you got to do.

Good luck

KurtE, 2019-10-25:
@OpusK,

Quick question - I again would like to remove constraint on how many servos that can be handled by different processors.   If we disregard the SyncRead/Write likewise Bulk...

One of the main things is a table that gets used to map an ID to a Model number, and my quick look through you are not really interested in the model number.  What instead is used is to differentiate some action depending on the type of servo and mostly this is within groups of servos:

And if you go to where these are actually used, they boil down to be used in different switch statements with groups of cases...   I sort gathered all of these and for each these I sort of marked groupings of the cases...
```
--------------------------------------------------------------------------
setBaudrate
--------------------------------------------------------------------------
AX12A: 0
AX12W: 0
AX18A: 0
DX113: 0
DX116: 0
DX117: 0
RX10: 0
RX24F: 0
RX28: 0
RX64: 0
EX106:     0
MX12W: 0
MX28: 0
MX64: 0
MX106: 0
MX28_2: 1
MX64_2: 1
MX106_2: 1
XL430_W250: 1
XC430_W150: 1
XC430_W240: 1
XXL430_W250: 1
XM430_W210: 1
XM430_W350: 1
XH430_V210: 1
XH430_V350: 1
XH430_W210: 1
XH430_W350: 1
XM540_W150: 1
XM540_W270: 1
XH540_W150: 1
XH540_W270: 1
XH540_V150: 1
XH540_V270: 1
PRO_M42_10_S260_R: 2
PRO_M54_40_S250_R: 2
PRO_M54_60_S250_R: 2
PRO_H42_20_S300_R: 2
PRO_H54_100_S500_R: 2
PRO_H54_200_S500_R: 2
PRO_M42_10_S260_RA: 3
PRO_M54_40_S250_RA: 3
PRO_M54_60_S250_RA: 3
PRO_H42_20_S300_RA: 3
PRO_H54_100_S500_RA: 3
PRO_H54_200_S500_RA: 3
PRO_H42P_020_S300_R: 3
PRO_H54P_100_S500_R: 3
PRO_H54P_200_S500_R: 3
PRO_M42P_010_S260_R: 3
PRO_M54P_040_S250_R: 3
PRO_M54P_060_S250_R: 3
---------------------------------------------------------------
bool Dynamixel2Arduino::setLedState(uint8_t id, bool state)
---------------------------------------------------------------
PRO_M42_10_S260_R:
PRO_M54_40_S250_R:
PRO_M54_60_S250_R:
PRO_H42_20_S300_R:
PRO_H54_100_S500_R:
PRO_H54_200_S500_R:
PRO_M42_10_S260_RA:
PRO_M54_40_S250_RA:
PRO_M54_60_S250_RA:
PRO_H42_20_S300_RA:
PRO_H54_100_S500_RA:
PRO_H54_200_S500_RA:
PRO_H42P_020_S300_R:
PRO_H54P_100_S500_R:
PRO_H54P_200_S500_R:
PRO_M42P_010_S260_R:
PRO_M54P_040_S250_R:
PRO_M54P_060_S250_R:
------------------------------------------------------------
bool Dynamixel2Arduino::setOperatingMode(uint8_t id, uint8_t mode)
------------------------------------------------------------
AX12A:  0
AX12W:  0
AX18A:  0
DX113:  0
DX116:  0
DX117:  0
RX10: 0
RX24F:  0
RX28: 0
RX64: 0

EX106: 1
XL320: 2
MX12W: 3
MX28: 3
MX64: 4
MX106: 4
MX28_2: 5
XL430_W250: 5
XC430_W150: 5
XC430_W240: 5
XXL430_W250: 5
MX64_2: 6
MX106_2: 6
XM430_W210: 6
XM430_W350: 6
XH430_V210: 6
XH430_V350: 6
XH430_W210: 6
XH430_W350: 6
XM540_W150: 6
XM540_W270: 6
XH540_W150: 6
XH540_W270: 6
XH540_V150: 6
XH540_V270: 6
PRO_M42_10_S260_R: 7
PRO_M54_40_S250_R: 7
PRO_M54_60_S250_R: 7
PRO_H42_20_S300_R: 7
PRO_H54_100_S500_R: 7
PRO_H54_200_S500_R: 7
PRO_M42_10_S260_RA: 8
PRO_M54_40_S250_RA: 8
PRO_M54_60_S250_RA: 8
PRO_H42_20_S300_RA: 8
PRO_H54_100_S500_RA: 8
PRO_H54_200_S500_RA: 8
PRO_H42P_020_S300_R: 8
PRO_H54P_100_S500_R: 8
PRO_H54P_200_S500_R: 8
PRO_M42P_010_S260_R: 8
PRO_M54P_040_S250_R: 8
PRO_M54P_060_S250_R: 8
-----------------------------------------------------------------------------
ControlTableItemInfo_t DYNAMIXEL::getControlTableItemInfo(uint16_t model_num, uint8_t control_item)
-----------------------------------------------------------------------------
AX12A: 0
AX12W: 0
AX18A: 0
DX113: 0
DX116: 0
DX117: 0
RX10: 0
RX24F: 0
RX28: 0
RX64: 0
EX106: 1
MX12W: 2
MX28: 2
MX64: 3
MX106: 4
MX28_2: 5
MX64_2: 6
MX106_2: 6
XL320: 7
XC430_W150: 8
XC430_W240: 8
XL430_W250: 8
XXL430_W250: 8
XM430_W210: 9
XM430_W350: 9
XH430_V210: 9
XH430_V350: 9
XH430_W210: 9
XH430_W350: 9

XM540_W150: 10
XM540_W270: 10
XH540_W150: 10
XH540_W270: 10
XH540_V150: 10
XH540_V270: 10
PRO_M42_10_S260_R: 11
PRO_M54_40_S250_R: 11
PRO_M54_60_S250_R: 11
PRO_H42_20_S300_R: 11
PRO_H54_100_S500_R: 11
PRO_H54_200_S500_R: 11

PRO_M42_10_S260_RA: 12
PRO_M54_40_S250_RA: 12
PRO_M54_60_S250_RA: 12
PRO_H42_20_S300_RA: 12
PRO_H54_100_S500_RA: 12
PRO_H54_200_S500_RA: 12
PRO_H42P_020_S300_R: 12
PRO_H54P_100_S500_R: 12
PRO_H54P_200_S500_R: 12
PRO_M42P_010_S260_R: 12
PRO_M54P_040_S250_R: 12
PRO_M54P_060_S250_R: 12
```
So far the biggest distinguishing list is the getControlTableIntemInfo, which appears to have 13 unique tables, now what I have not yet done is to see if some of the other tables conflict.  LIke some servo being in one grouping in one operation and a different grouping in a different operation...

What I am wondering is instead of keeping the model number for each servo, we could keep a grouping index number.   If it turns out less < 16 could save this info in 4 bits... So to hold information for 254 servos would take 127 bytes.  Reserve value 0 maybe as unknown/not in list... 

This could allow for much faster mapping.  If you want to know which type servo 23 is, you don't have to linear search a list to see if it is in the list and then grab value.  Instead you simply compute that this is in byte 12... and grab it...  

Also the places that use this like setOperatingMode or the functions that use getControlTableItemInfo would have a much short list of items to use in switch statement and/or could have simple array to load data from... 

But before I consider trying this out, does this sound reasonable to you and/or other issues you can think of.

Thanks
Kurt

OpusK, 2019-10-28:
Hi, @KurtE 

I've been on a short vacation last Friday.
Thank you for your feedback and comments.

First of all, when it comes to converting c++ to c, I thought about it because there was a request from the company.

I agree with many of the things you mentioned.
However, for users who only wanted to use c, they had to do a minimal amount of work.
So, first of all, I decided to implement c only for protocols. (protocol.cpp -> protocol.c)

This work is in progress in the branch below. (I think protocol.c is done)
- [code_layer_redesign](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/tree/code_layer_redesign)

I used this to rewrite master and slave.
`ping()` has changed parameters.
As for sync and bulk, I haven't implemented it yet because I want to talk more with you.
In particular, since the API of Dynamixel2Arduino Class has not changed, the existing examples are not a problem. However, it affects raw examples using the Master function. (as for sync, bulk, ping ...)

And, for packet buffer allocation, I applied your dynamic allocation.
The class is now internally initially allocating a buffer with a size of DXL_BUF_MAX, but I'm also considering how to determine it through class parameters. (Or create a static buffer using a template ...)

Anyway, could you please review the changes?

For your second article, I will read and understand it more and then reply to you.

Thanks again for your contribution!

OpusK, 2019-10-28:
@KurtE 

> And if you go to where these are actually used, they boil down to be used in different switch statements with groups of cases... I sort gathered all of these and for each these I sort of marked groupings of the cases...

I've considered this before, but I've put it on hold for scalability.

> What I am wondering is instead of keeping the model number for each servo, we could keep a grouping index number. If it turns out less < 16 could save this info in 4 bits... So to hold information for 254 servos would take 127 bytes. Reserve value 0 maybe as unknown/not in list...

Even if we can reduce it to 4 bits, if we can't reasonably use the additional 4 bits, it looks good to use all 8 bits. BTW, it would be nice to use an 8-bit index for model numbers represented in 16-bit.

> Also the places that use this like setOperatingMode or the functions that use getControlTableItemInfo would have a much short list of items to use in switch statement and/or could have simple array to load data from...

It also looks good with arrays instead of switch statements.

KurtE, 2019-10-29:
Hi @OpusK,

I meant to reply yesterday, but it was lost...  So will try again. 

Will take a look at your new branch hopefully in the next few days.  I am doing some stuff with Teensy, and was going through and updating some of my current code to use this library, to get ideas of stuff, that I like or don't like.  

Have another branch of my fork (https://github.com/KurtE/Dynamixel2Arduino/tree/develop_allow_all_ids) that I was doing some hacking on to allow all IDS... So far it keeps a const PROGMEM table of all currently defined dynamixel types and uses an array of indexes to access these tables.  So currently taking 254 bytes of storage, plus program space of  Servos: 

Am thinking about maybe modifying to either:
a) Maybe modify to only allow up to N unique servo types (example 16, actually 15 as you reserve a value like 0xf to say not used....  Then a 4 bit field would work fine.  So could cut the big table in half. 
and use a couple simple function/macros to extract/set the servo_type index sort of something like:
servo_type_index = (id & 1)? servo_index_list[id >> 1) >> 4 : servo_index_list[id >> 1) & 0xf;
And then use that value to retrieve the model from your table of up to 15 values...   This does nothing for reducing your case statements in the code for models. 

b) instead of keeping list of model numbers, we simply define model_groupings, and keep index like above.  If we can reduce again down to 15 groups can use same type of calculations and cut in half.

Side note: we could use different number of bits per index other than 4 or 8, it can make the calculations a little more complex although in some cases we can waste a little memory and make it easier still.  Like  if 15 is too tight, so we wish to go up to 31 groupings.   
In this case I might waste a few bits: 

(8 bit)Again currently I have: 254 bytes for a byte index (max 255 types)
(4 bit)This could be cut down to 127 bytes for nibble index (max 15 types)
(5 bit) I would probably pack 3 into 16 bits: So could use 159 bytes but I would probably use170 
bytes

The 5 bit math is not quite as easy but in logic it would be:
```
switch (id%3) {
  case 0: return servo_index_list[id/3) & 0x1f;
  case 1: return (servo_index_list[id/3)>>5) & ox1f;
  default: return  (servo_index_list[id/3)>>10;
}
```
Again we could do it and save the extra few wasted bits, but maybe not worth it... 

Got to run... Again will try to take look through your current stuff soon. 

Kurt

OpusK, 2019-10-31:
> Wondering: if about if you should or want some form of standardized naming convention for structures and functions?
That is with C++ you can and were using the namespace feature to help make sure that whatever you define should not conflict with other sub-systems...

First, I used prefixes like `DXL_` because I couldn't use the CPP namespace and I had to use C only.
In addition, I had to give a name that would not conflict with DYNAMIXEL for users who want to use it with the DYNAMIXEL SDK at the same time.

Even when using the CPP namespace, `DYNAMIXEL::PortHandler` conflicts with `dynamixel::PortHandler` in DynamixelSDK.
I thought about changing the namespace, but I just used the name change method.

And protocol.c is intended to be used independently of Arduino. Therefore, it did not follow Arduino's coding style, but rather ROBOTIS's internal coding style.

> Part of the ability to sub-class the Port handler code may only be needed for maybe three virtual things (or set-able callback functions):

Do you want the virtual function of the class to be created as shown below?

```cpp
class SerialPortHandler : public DXLPortHandler
{
  public:
    SerialPortHandler(HardwareSerial& port, const int dir_pin = -1);

    virtual void begin() override;
    virtual void end() override;
    virtual int available(void) override;
    virtual int read() override;
    virtual size_t write(uint8_t) override;
    virtual size_t write(uint8_t *buf, size_t len) override;

    virtual void begin(unsigned long baud);
    virtual unsigned long getBaud() const;
    virtual void setTxMode();
    virtual void setRxMode();

  private:
    HardwareSerial& port_;
    const int dir_pin_;
    unsigned long baud_;
};
```

OpusK, 2019-10-31:
@KurtE 

I referred to the ModelNumber management you designed and applied it using the changed protocol.h API.
I would appreciate it if you review it.

KurtE, 2019-10-31:
@OpusK  - Good Morning, I will review the model number stuff 

Naming conventions: Sounds like a difficult one for you as there are many different clients on many different platforms with different conventions.   I probably have it a bit easier as I only have to satisfy myself :D 

DXLPortHandler definitions... Yes you don't need an additional method for init as begin is fine.  I don't like to do much in constructors, as in c++ you may not have much control on order of when the different constructors are called, like even like is the underlying hardware object like the Serial port initialized at all and in some case could fault out...

But if it were me, setting up the interfaces, I would probably change the definitions slightly.   I would probably start off changing DXLPortHandler, to be derived from the Arduino stream class, which is derived from the Print class
```
class DXLPortHandler : public stream
{
  public:
    DXLPortHandler();
    
    virtual void begin() = 0;
    virtual void end() = 0;
    virtual int available(void) = 0;
    virtual int read() = 0;
    virtual size_t write(uint8_t) = 0;
    virtual size_t write(uint8_t *buf, size_t len) = 0;
    bool getOpenState();
    void setOpenState(bool);

  private:
    bool open_state_;
};
```
I believe most if not all of your virtual functions are already coming from these anyway. 
You then have the option to use any of the standard underlying stream and print functions without having to code them, but your classes must provide at minimum a read/write function.

As for setTXMode, setRxMode, there are a few different options, depending on how you might organize your code. 
a) Current way and not need them.  Currently your code puts them in forcing in and out of write mode for every call of write.  So if your top level code is wanting to call these only once per packet, then it does not matter if they are separated out or not. 

b) One new member like setTxRxMode(bool do_write); 
c) Two members like you mentioned. 
Where instead of having write wait for each one to complete before you return, you have the calling code, tell the port code, that it is done outputting and should switch back to RX mode.   Where this can be handy is suppose you wish to output something like a SyncWrite,  currently I believe you still generate the whole packet in a buffer to send off at once.   Alternatively one could setup the code, to call: the write of the header information, then loop through outputting the data for each servo that you are outputting data for (while still computing stuff like checksum), and finally output the packet ending data (checksum) and at this point tell system I am done outputting.   Advantage here, is you don't have to have a large buffer to put all of this stuff in, and you also allow the Serial port to start to output data while you are still putting stuff out in the queue... 

Hope that makes sense?   Either way works. 

Will get back later on model number...

KurtE, 2019-10-31:
@OpusK, 

Not sure if best to edit previous comment or put in new one... 

On 
`Master::ping(uint8_t id, uint8_t *p_recv_buf, uint8_t recv_buf_capacity, uint32_t timeout_ms)`
in the broadcast case, you currently are throwing away the other information returned by the servos.

So you are only returning one byte per servo....  Which is consistent with your top level ping function:
`bool Dynamixel2Arduino::ping(uint8_t id)`
Where you pass in: 
`recv_cnt = Master::ping(DXL_BROADCAST_ID, recv_ids, sizeof(recv_ids), 3*253);`

Which of course this currently busting my simple test app, which finds servos... 
[OpenCM_Find_Servos-191031a.zip](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/files/3794736/OpenCM_Find_Servos-191031a.zip)

Again nothing special, here but I did have: 
```
XelInfoFromPing_t ping_info[32];
void FindServos(void) {
...
      Serial.println("  Try Protocol 2 - broadcast ping: ");
      Serial.flush(); // flush it as ping may take awhile... 
      
      if (uint8_t count_pinged = dxl.ping(DXL_BROADCAST_ID, ping_info, 
            sizeof(ping_info)/sizeof(ping_info[0]))) {
        Serial.print("Detected Dynamixel : \n");
        for (int i = 0; i < count_pinged; i++)
        {
          Serial.print("    ");
          Serial.print(ping_info[i].id, DEC);
          Serial.print(", Model:");
          Serial.print(ping_info[i].model_number, HEX);
          Serial.print(", Ver:");
          Serial.println(ping_info[i].firmware_version, HEX);
          g_servo_protocol[i] = 2;
        }
      } else Serial.printf("Broadcast returned no items(%x)\n", dxl.getLastLibErrCode());
...
```
Which fails to compile.  In the Protocol 2 case having the ability to get all of the information in one shot is nice.   Assuming you still support this type of Ping, might not be too bad an idea to change the name of the structure as what is XEL?  (Yes I know...)

Kurt

OpusK, 2019-11-01:
Hi, @KurtE 

It's quite messy with a lot of issues. But the title is Wishlist so I think it is fine :)
I hope you are comfortable.

> But if it were me, setting up the interfaces, I would probably change the definitions slightly. I would probably start off changing DXLPortHandler, to be derived from the Arduino stream class, which is derived from the Print class

Probably, we want to make it available on platforms other than Arduino for Master / Slave / PortHandler.
Therefore, I thought `DXLPortHandler` should be generic.
So in my opinion, when creating a handler for Arduino like `DYNAMIXEL::SerialPortHandler`,
I think it's good to add features.

> `Master::ping(uint8_t id, uint8_t *p_recv_buf, uint8_t recv_buf_capacity, uint32_t timeout_ms)`
in the broadcast case, you currently are throwing away the other information returned by the servos.

The reason I made this change was to reduce memory usage.
Probably, we can use the structure method we provided earlier, and we can provide an overloaded function for this. (It seems to be the level to revert the previous function.)

> So I believe in both Protocol 1 and Protocol 2 here it is going to send/receive two sets of packets.
First a ping packet and wait for response, if it gets a response, it then sends off a packet associated with getModelNumber(id), which will do a read of the model number or 0xffff it fails.

`Dynamixel2Arduino` needs to know the model number correctly because of its API characteristics.
If the user tries to operate the motor without executing the ping command, there will be a problem because there is no information in the table.
Therefore, when retrieving information from a table, if no valid information exists, the ping command was applied.
However, since it doesn't match the API name and ping packets can be sent differently from the user's intention, it is better to remove it.
(However, the user must execute the ping command, and we must notify it.)

OpusK, 2019-11-01:
@KurtE 

In regards to the buffer configuration, I am considering whether Master will keep the default malloc and the buffer replaceable, Dynamixel2Arduino as the Template class, and finally pass the static buffer to the Master class.

```cpp
template <uint8_t DXL_MAX_NODE = 16,
                  uint16_t MAX_DATA_SIZE_PER_NODE = 8>
class Dynamixel2Arduino_ : public DYNAMIXEL::Master;

or

template <uint16_t PACKET_BUF_SIZE = DEFAULT_DXL_BUF_LENGTH>
class Dynamixel2Arduino_ : public DYNAMIXEL::Master;
```

This can also be used in the form(like [rosserial](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros_lib/ros.h)):

```cpp
 #if defined (ARDUINO_AVR_UNO) || defined (ARDUINO_AVR_YUN) \
   || defined (ARDUINO_AVR_INDUSTRIAL101)
  typedef Dynamixel2Arduino_ <16, 8> Dynamixel2Arduino;
 #elif defined (ARDUINO_AVR_LEONARDO)
  typedef Dynamixel2Arduino_ <16, 12> Dynamixel2Arduino;
 #elif defined (OpenCR)
  typedef Dynamixel2Arduino_ <254, 32> Dynamixel2Arduino;
 #else
  typedef Dynamixel2Arduino_ <32, 16> Dynamixel2Arduino;
 #endif
```

What do you think about this?

OpusK, 2019-11-01:
@KurtE 

In https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/commit/a4e4821fed85655c959a2331de20bb2c8156454b, I reverted one ping function.

However, I'm still worried about applying this to the Dynamixel2Arduino's ping in terms of memory. (May apply for single pings)
I think it's applicable for a single ping (the structure is 4 bytes long), but for broadcasting, the current 254 bytes are now growing to 1016 bytes.

KurtE, 2019-11-01:
@OpusK 

Quick question: I am not seeing anywhere right now that is filling in the model_number_table?
I did a global search in my downloaded fork of your code and only saw:
```
Searching 95 files for "model_number_table"

D:\GitHub\Dynamixel2Arduino\src\Dynamixel2Arduino.cpp:
   21  namespace DYNAMIXEL{
   22  
   23: const uint16_t model_number_table[] PROGMEM = {
   24      AX12A, AX12W, AX18A,
   25      
   ..
   55  };
   56  
   57: const uint8_t model_number_table_count = sizeof(model_number_table)/sizeof(model_number_table[0]);
   58  
   59  enum Functions{
   ..
  822      ret = model_number_idx_last_index_;
  823    }else{
  824:     for(i=0; i<model_number_table_count; i++)
  825      {
  826:       if(model_num == pgm_read_word(&model_number_table[i])){
  827          model_number_idx_last_index_ = i;
  828          ret = i;
  ...
  847  
  848    idx = model_number_idx_[id];
  849:   model_num = (idx < model_number_table_count) ? pgm_read_word(&model_number_table[idx]) : UNREGISTERED_MODEL;
  850  
  851    return model_num;

8 matches in 1 file
```
It was being set in the getModelNumberFromTable by using the ping function.  I know I sort of had issue with calling ping then, but what I meant to say was something like instead of ping, we could call... 

Maybe now something like: 
```
uint16_t Dynamixel2Arduino::getModelNumberFromTable(uint8_t id)
{
  uint8_t idx;
  uint16_t model_num;

  if(id > 254){
    setLastLibErrCode(DXL_LIB_ERROR_INVAILD_ID);
    return UNREGISTERED_MODEL;
  }

  idx = model_number_idx_[id];
  if (idx >= model_number_table_count) {
     idx =  getModelNumberIndex(getModelNumber(id));  // send out the one query. 
     model_number_idx_[id] = idx;
  }
  model_num = (idx < model_number_table_count) ? pgm_read_word(&model_number_table[idx]) : UNREGISTERED_MODEL;

  return model_num;
}
```

I was also trying to say that the code should try to be opportunistic when possible.   
That is  if you are doing a ping on a Protocol 2 servo and internally it returns the full  3 bytes of data including the model number, you should extract that data and update the model_number_idx_ table.   As to save you from later having to query this data later. 

Thanks for putting back in the full ping function.   Unless there is somewhere in the underlying code that calls this with a full 253x3 byte setup, I am not as worried about it's memory usage, as the application should use it in a reasonable way.  That is for example for my hexapods, In most cases I don't expect more than maybe 6*3+2= 20 servos or on the 4dof version 6*4+2 = 26 servos.  So I am never likely to going to call it to return more than information > 20 to 30 servos... 

As for template stuff, I have not used them much, but if it works for you, great.  I prefer the version:
```
template <uint16_t PACKET_BUF_SIZE = DEFAULT_DXL_BUF_LENGTH>
class Dynamixel2Arduino_ : public DYNAMIXEL::Master;

//private member
uint8_t packet_buf[PACKET_BUF_SIZE ];
```
Note I edited it slightly as pakcet_buf -> packet_buf

But which form of this may depend on for example how you do the sync read/write... If you still have it where you have a structure setup with the max data per node, in the structure and MAX ID count of these, than your first one makes sense.

OpusK, 2019-11-04:
Hi, @KurtE 

> I know I sort of had issue with calling ping then, but what I meant to say was something like instead of ping, we could call...

Do you want to register without using `ping` like below?

```cpp
bool setModelNumber(uint8_t id, uint16_t model_number);
```

> I am not as worried about it's memory usage, as the application should use it in a reasonable way. 

What I said was about the `ping()` function in the `Dynamixel2Arduino` class.
Anyway, as you suggested for `single ping()`, it would be better to get model information specific to protocol2.

In the case of broadcast ping (), I'm going to take the form of double-fetching the information about the ID and the model. If you don't want this, you can use `master's ping()` function, as you say.

> As for template stuff,

This has a number of considerations, so I'm going to keep using dynamic allocation as I do now. It may be considered again later.

Now I'm thinking about how to express the syncRead / Write function. As you know, the protocol's packet function is a form that can continue to add parameters, so it's easier to use the way you made it into a class. Here's the form of the API I thought about last week (although I haven't defined it yet ...) What do you think about this?
 - [APIs](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/code_layer_redesign/src/utility/master.h#L95)
```cpp
    bool syncRead(uint8_t *p_param, uint16_t param_len, uint8_t id_cnt,
      uint8_t *p_recv_buf, uint16_t recv_buf_capacity, uint32_t timeout_ms = 10);
    bool syncWrite(uint8_t *p_param, uint16_t param_len, uint32_t timeout_ms = 10);
    bool bulkRead(uint8_t *p_param, uint16_t param_len, uint8_t id_cnt,
      uint8_t *p_recv_buf, uint16_t recv_buf_capacity, uint32_t timeout_ms = 10);
    bool bulkWrite(uint8_t *p_param, uint16_t param_len, uint32_t timeout_ms = 10);

    /* Easy functions for Sync Read */
    bool beginSyncRead(uint16_t addr, uint16_t addr_len);
    bool addSyncReadID(uint8_t id);
    bool sendSyncRead(uint8_t *p_recv_buf, uint16_t recv_buf_capacity);

    /* Easy functions for Sync Write */
    bool beginSyncWrite(uint16_t addr, uint16_t addr_len);
    bool addSyncWriteData(uint8_t id, uint8_t *p_data, uint16_t data_len);
    bool sendSyncWrite();

    /* Easy functions for Bulk Read */
    bool beginBulkRead();
    bool addBulkReadID(uint8_t id, uint16_t addr, uint16_t addr_len);
    bool sendBulkRead(uint8_t *p_recv_buf, uint16_t recv_buf_capacity);

    /* Easy functions for Bulk Write */
    bool beginBulkWrite();
    bool addBulkWriteData(uint8_t id, uint16_t addr, uint8_t *p_data, uint16_t data_len);
    bool sendBulkWrite();
```

As you suggested (as workbench and sdk did), would it be better to support separate classes?

KurtE, 2019-11-04:
@OpusK ,  I will take a more detailed look at API, but initial look, looks reasonable.  As for adding classes?  Not sure it helps or not.  Sometimes makes things look a bit cleaner if it holds all of the book keeping like buffers and the like but...

Ping:
If you do a ping and it returns model information, you should for sure take advantage of it and update the model table.  

All I was trying to say earlier is If the user does something like:  dxl.setGoalPosition(1, 1024);
And you don't yet know what servo 1 model number is.  The current code might call ping, to get that information, which is fine if you are currently running Protocol 2, as ping will return that data, which you can then internally populate the model table.  However for example if you are running Protocol 1, you still call ping, and if the ping succeeds it then calls indirectly through Read to red the model number from the standard control table index and then fills in the data.  In this case we can bypass calling ping and instead do the read directly so it only needs one DXL operation...

OpusK, 2019-11-05:
@KurtE 

> The current code might call ping, to get that information, which is fine if you are currently running Protocol 2, as ping will return that data, which you can then internally populate the model [...]

You are talking about `ping() in Master`, not `ping() in Dynamixel2Arduino` class?

Prior to this update, as you said, the library included code to handle internally when calling command functions without using `ping()`.
I think this method is good when you consider the characteristics of Dynamixel2Arduino. However, the previous code did not match the name of the function because it called `ping()` from `getModelNumberFromTable()`.

Therefore, I want to support that function with the following code.

```cpp
float Dynamixel2Arduino::readForRangeDependencyFunc(uint8_t func_idx, uint8_t id, uint8_t unit)
{
  float ret = 0;
  int32_t recv_data = 0;
  uint16_t model_num = getModelNumberFromTable(id);
  ItemAndRangeInfo_t item_info;

  // To use the command function without ping() or model addition.
  if(model_num == UNREGISTERED_MODEL){
    if(ping(id) == true){
      model_num = getModelNumberFromTable(id);
    }
  }

  if(model_num != UNREGISTERED_MODEL){
    item_info = getModelDependencyFuncInfo(model_num, func_idx);

    if(item_info.item_idx != LAST_DUMMY_ITEM){
      recv_data = readControlTableItem(model_num, item_info.item_idx, id);
      checkAndconvertReadData(recv_data, ret, unit, item_info);
    }else{
      setLastLibErrCode(D2A_LIB_ERROR_NOT_SUPPORT_FUNCTION);
    }    
  }else{
    setLastLibErrCode(D2A_LIB_ERROR_UNKNOWN_MODEL_NUMBER);
  }

  return ret;
}

bool Dynamixel2Arduino::writeForRangeDependencyFunc(uint8_t func_idx, uint8_t id, float value, uint8_t unit)
{
  bool ret = false;
  int32_t data = 0;
  uint16_t model_num = getModelNumberFromTable(id);
  ItemAndRangeInfo_t item_info;

  // To use the command function without ping() or model addition.
  if(model_num == UNREGISTERED_MODEL){
    if(ping(id) == true){
      model_num = getModelNumberFromTable(id);
    }
  }

  if(model_num != UNREGISTERED_MODEL){
    item_info = getModelDependencyFuncInfo(model_num, func_idx);

    if(item_info.item_idx == LAST_DUMMY_ITEM)
      return false;

    if(checkAndconvertWriteData(value, data, unit, item_info) == false)
      return false;

    ret = writeControlTableItem(model_num, item_info.item_idx, id, data);
  }

  return ret;
}
```

KurtE, 2019-11-09:
Hi @OpusK - sorry been off doing some other stuff, 

Need to figure out what all has changed.  I have my fork/branch I believe in sync with yours, and my simple find servos program, which looks at both ports plus Protocol 1 and 2 on each and tries to simple ping and then on protocol 2, more complex ping... 

Earlier it was finding my servos, now it is not... So probably some functional difference I need to figure out and adapt...   Once this is working again I will work on porting over my Phoenix code (Hexapod) to use this library instead of DynamixelSDK.. 

Now back to playing

OpusK, 2019-11-11:
Hi, @KurtE 

> sorry been off doing some other stuff,

You don't have to be sorry for me :)
I hope you enjoy and give me feedback when you want.
I am always grateful for your contribution.

For the sync functions, it's going to take a bit longer because of my other things.

KurtE, 2019-11-12:
Hi @OpusK, 

I have tried my simple program to find servos.  Currently on the buss there are about 12 protocol 2 servos, and it is not finding any of them. 

I tracked it down in the PING code, it properly sends the packets and then calls rxStatusPacket which is failing to return the data.   The servos are  responding and you are parsing the data... 

I have updated this code to dump some data: 
```
const InfoToParseDXLPacket_t* 
Master::rxStatusPacket(uint8_t* p_param_buf, uint16_t param_buf_cap, uint32_t timeout_ms)
{
  InfoToParseDXLPacket_t *p_ret = nullptr;
  DXLLibErrorCode_t err = DXL_LIB_OK;
  uint32_t pre_time_ms;

  // Parameter exception handling
  if(p_port_ == nullptr
  || (param_buf_cap > 0 && p_param_buf == nullptr)){
    err = DXL_LIB_ERROR_NULLPTR;
  }else if(p_port_->getOpenState() != true){
    err = DXL_LIB_ERROR_PORT_NOT_OPEN;
  }
  if(err != DXL_LIB_OK){
    last_lib_err_ = err;
    return nullptr;
  }

  // Receive Status Packet
  begin_parse_dxl_packet(&info_rx_packet_, protocol_ver_idx_, p_param_buf, param_buf_cap);
  pre_time_ms = millis();
  while(1) 
  {
    if(p_port_->available() > 0){
      uint8_t new_data = p_port_->read();
      Serial.printf("parse %x %x ", new_data, info_rx_packet_.parse_state);
      err = parse_dxl_packet(&info_rx_packet_, new_data);
      Serial.printf(" -> %x %x %x\n",info_rx_packet_.parse_state, info_rx_packet_.err_idx, err);
      if(err == DXL_LIB_OK){
        Serial.printf("** Parse shows DXL_LIB_OK status: Protocol %d %x %x\n", protocol_ver_idx_, info_rx_packet_.inst_idx, info_rx_packet_.err_idx);
        if((protocol_ver_idx_ == 2 && info_rx_packet_.inst_idx == DXL_INST_STATUS)
        || protocol_ver_idx_ == 1){
          if(info_rx_packet_.err_idx == 0){
            p_ret = &info_rx_packet_;
          }
          break;
        }
      }else if(err != DXL_LIB_PROCEEDING){
        break;
      }
    }

    if (millis()-pre_time_ms >= timeout_ms) {
      err = DXL_LIB_ERROR_TIMEOUT;
      break;
    }
  }

  last_lib_err_ = err;

  return p_ret;
}
```
So for example when it pings servo 2, I am getting the following data:
```
CM9.04 Find Servos program
*** OpenCM9.04 setDxlMode ***
Setting Baud to: 1000000

Search for all servos at baud rate: 1000000
Begin Searching on Port: 1
1.00
  Begin Protocol 1: 
  Done
Begin Searching on Port: 1
2.00
  Begin Protocol 2 Simple Ping: 
parse ff 0  -> 0 0 1
parse ff 0  -> 0 0 1
parse fd 0  -> 1 0 1
parse 0 1  -> 2 0 1
parse 2 2  -> 3 0 1
parse 7 3  -> 4 0 1
parse 0 4  -> 5 0 1
parse 55 5  -> 6 0 1
parse 80 6  -> 7 80 1
parse 24 7  -> 7 80 1
parse 4 7  -> 7 80 1
parse 29 7  -> 8 80 1
parse da 8  -> 9 80 1
parse ef 9  -> 0 80 0
** Parse shows DXL_LIB_OK status: Protocol 2 55 80
...
So it properly parses the packet and returns with err=0. 
However it fails to return any data as for the check: if(info_rx_packet_.err_idx == 0){
Which from the previous output shows a value of 0x80, which looks like it set when it processed the 0x80 byte...  Which I understand is the HW error.   I sometimes get this when I power up the servos depending on the order things power up. 

In some sketches like my SyncReadWrite sketch, I have code that detects these errors and if found, tells that servo(s) to reboot (dxl.reboot()) to clear it. 

So question is, do you really want all of these rxStatusPacket messages to fail if the servo returns an alert status (bit 7 set) ? 

Kurt

```

OpusK, 2019-11-13:
Hi, @KurtE 

> So question is, do you really want all of these rxStatusPacket messages to fail if the servo returns an alert status (bit 7 set) ? 

There are still many ways to deal with packet errors.
This is a policy issue, and a packet error means that the packet communication itself was successful. So I'm wondering if it's better to return it as true in rxStatusPacket and false in each instruction function (ping, read/write, etc.).

Would you please let me know if you have any thoughts on this?

KurtE, 2019-11-13:
@OpusK, 
I think it may depend on the type of error. 

That is there are errors, where the program is at fault.  that is you send an invalid op code, or ask for a register that is out of range... These I would expect to show the failed state.   

There are other conditions like, Alerts (where the Servo has a condition set, like it was under powered or over powered... where the high bit is set, that I believe in many conditions the operation may have succeeded?  Example Ping.  I suppose you could have them fail, but then you may need each time to ask the system, why?  If the alert status, than you may need to query to find out the reason?  But if you fail out on the error condition, then the register read will fail, so you won't get this information.  

Again we probably may want to look through to see how different libraries currently return the data or error out in conditions like this.

OpusK, 2019-11-13:
@KurtE 

When I asked the DYNAMIXEL team, they said that hardware errors occur separately from packet errors.
Therefore, unless a packet error occurs, even if a hardware error occurs, the control table can be accessed normally (value changed or read).

Therefore, the hardware error is not a packet error, so the packet processing function considered the packet processing successful.

However, as you said, there are a number of ways of returning errors.
Some libraries divide errors, warnings, and levels into different ways of dealing with each.

Currently, D2A simply returns true, false, and adopts a separate function (getLastLibErrCode (), getLastStatusPacketError ()).

But I'm feeling sorry for this approach, and I keep thinking about it. (I'm busy with other things right now so I don't have much time to worry ...)

> Again we probably may want to look through to see how different libraries currently return the data or error out in conditions like this.

Anyway, I totally agree with you.

KurtE, 2019-11-13:
@OpusK, 

Thanks, I did a sync up to your changes, which was similar to mine 
Although in my case I only fixed for Protocol 2... 

As for Protocol 1: it would be interesting to figure out which of the bits should be considered program errors versus hardware issue. 

I would probably consider (Overload, Overheating and Input voltage) as hardware issue 
And (Instruction, checksum, probably Range and Angle limit) as program errors... 

But some of these are guesses, like Angle limit?  I am not sure in what conditions this happens.  
I can imagine if I tell an AX servo to go to position 2000 it would give me this error.  But if the servo happens to be in the 300-360 degree area at startup and I query for present position does it return this error?  then it is more of a hardware condition.  But again I did not test this to see...

KurtE, 2019-11-13:
@OpusK, 

I thought I would play with my Sync read and write c++ code, and running into some issues. 
Some of the api names changed like something like txPacketInst went to txInstPacket

Also things like: DXL_INST_SYNC_WRITE
Is no longer in a names space, so things like: DYNAMIXEL::DXL_INST_SYNC_WRITE
does not compile

Maybe by design, but thought I would mention

OpusK, 2019-11-14:
@KurtE 

> I would probably consider (Overload, Overheating and Input voltage) as hardware issue
And (Instruction, checksum, probably Range and Angle limit) as program errors...

I'll look at how packet processing is done internally.

> I thought I would play with my Sync read and write c++ code, and running into some issues.
Some of the api names changed like something like txPacketInst went to txInstPacket
Also things like: DXL_INST_SYNC_WRITE
Is no longer in a names space, so things like: DYNAMIXEL::DXL_INST_SYNC_WRITE
does not compile

Since it has not been released yet, some APIs have been renamed, and the protocol has been written in c and no namespace is available. I'm sorry for any inconvenience this may cause without telling you about these changes.

KurtE, 2019-11-17:
@OpusK,

Quick update: I had some time a day or two ago, and I fixed my test app that did a sync read and write of the 18 servo hexapod,  using my own C++ Sync Read and Write classes I mentioned earlier.

So now I am looking to convert my C code over to your new APIs.  Starting first with a version of Sync Read.  I find it much easier to understand things like proposed APIS, when I actually try to use them. 

So I am trying to figure out in practice how I would use these new API's, which I grabbed from your earlier post.  Here is just the SyncRead:
```
bool syncRead(uint8_t *p_param, uint16_t param_len, uint8_t id_cnt,
              uint8_t *p_recv_buf, uint16_t recv_buf_capacity, uint32_t timeout_ms = 10);
/* Easy functions for Sync Read */
bool beginSyncRead(uint16_t addr, uint16_t addr_len);
bool addSyncReadID(uint8_t id);
bool sendSyncRead(uint8_t *p_recv_buf, uint16_t recv_buf_capacity);
```
Suppose, my Robot has 18 servos:
`const uint8_t ServoIDList[] = {19, 3, 5, 2, 4, 6, 7, 9, 11, 8, 10, 12, 13, 15, 17, 14, 16, 18};`
 Three per leg (Coxa, Femur, Tibia) 

**Warning Side track:** he Original AX version of the hexapod originally had servo 1 instead of 19, BUT had/have issue that from time to time a servo will reset to have ID#1.  So hexapod code detects this if it finds all of the expected servos except one AND servo #1 exists, then it renumbers 1 to the missing number... 

Now suppose in a main loop, where maybe I wish to query these 18 servos maybe 20 or 50 times per second, what do I do? 

Top level API?  What is param and param_len?  Is the actual data sent out in the Sync Read?  
```
#define DXL_X_PRESENT_POSITION     132 // 4 (R)
const uint8_t sync_read_pos[] = {
        DXL_X_PRESENT_POSITION & 0xff, DXL_X_PRESENT_POSITION >> 8, 4, 0, 
        19, 3, 5, 2, 4, 6, 7, 9, 11, 8, 10, 12, 13, 15, 17, 14, 16, 18};
uint8_t sync_recv_buffer[200];
```
As this is only supported for Protocol 2, not an issue about differences between protocol 1 and 2.

**Warning side track**: Unless you are maybe using some other Servo Controller like USB2AX, who had the controller emulate the Sync Read,  But their return format was different, as they then did individual query to the specified servos and then collected the data into one packet which they sent back?  Why to minimize USB delays... (End side track)

If the above input is correct, are you going to support two versions of this?  where on AVR maybe it is const PROGMEM?   does the call to syncRead modify the p_param data?  if not shouldn't be marked as const?

So now suppose I call off to Sync read like:
`if (syncRead(sync_read_pos, sizeof(sync_read_pos), sync_recv_buffer, sizeof(sync_recv_buffer)) {...}`
What is returned?  Suppose one of the servos did not respond?  would it fail?  How do I find out how many responded? 

I am guessing that the data in sync_recv_buffer, will be in a format, like: servo id#, data for that ID, servod id#... ? 

Or How do you see using the Easy functions?  Do you expect that in the main loop I would do something like:
```
void loop() {
    ...
    beginSyncRead(DXL_X_PRESENT_POSITION, 4);
    for (uint8_t i=0; i < sizeof(ServoIDList); i++) addSyncReadID(ServoIDList[i]);
    if (sendSyncRead(sync_recv_buffer, sizeof(sync_recv_buffer)) {.... }
...
```
That is there is only one Sync Read setup?  Of do you see the initial part going into setup()?  Again that would restrict you to one sync read for your sketch? 

Or do you maybe see some other API, that allows you to save away the query into a buffer that you can then use the top level API form?   Again maybe not that helpful as in the sync write case the ID and data are interspersed. 

Also do you see another helper function(s), that helps the user get the data from the returned sync read data buffer?  i.e. maybe something like: 
```
// retrieve buffer?
bool retrieveSyncReadData(uint8_t id, uint8_t *id_buffer, uint16_t id_buffer_size); 
```
Might need to pass in previous buffer pointer as well?  Or remember it?  

What about different processors that may have different byte orders?  little endian versus big endian?
Do you see helper functions, like the retrieve, by size?   Sort of like:
```
uint16_t retrieveSyncReadData2(uint8_t id, uint16_t offset=0);
uint32_t retrieveSyncReadData4(uint8_t id, uint16_t offset=0)
uint32_t retrieveSyncReadDataSize(uint8_t id, uint8_t size, uint16_t offset=0);
```
And if you go to this extent, do you then go overboard and also have logical register names, which look up in your by servo type the actual parameter number for that ID and size? Sort of like you have the workbench?  Again maybe overkill,  but you may still need to think about how the endian stuff impacts the library, especially when you are talking about different processors, not just Arduino... 

syncWrite:  will not  say much more yet, will wait to see how syncRead flushes out. 

But again, this can be a little more interesting with sync write, as it is supported on both Protocol 1 and protocol 2, and their data will be different, as starting register number is 1 or 2 bytes and likewise count.  let alone register numbers and sizes may very likely be different. 

Again hopefully I did not ramble too much or get too far off track.

Kurt

KurtE, 2019-11-17:
@OpusK 

Sorry, looks like maybe I am jumping the gun to see about writing up a C implementation of SyncRead?

That is so far I am not seeing any functions/methods defined to transfer and receive packets using C.

I am hoping that you would still have a simple method, like: 
`	if (!DXLTxPacketInst(DXL_BROADCAST_ID, INST_SYNC_READ, (void*)&SRData->starting_addr,  SRData->cnt_servos + 4)) return false; // failed on tx`

That is some C function , that maybe takes the servo ID, command, pointer to data, and size of the data. 

As I mentioned in previous post, that maybe you would not need to copy the data into some other buffer, but could potentially break up the output of the packet into chunks:  Probably 2 or 3 chunks.
first chunk would have all of the header data.  2nd chunk would be extra data, and 3rd chunk would be computed checksum... 

That is sort of how I did it in the past, where I had functions like: setTXMode(), then do the output of the three cunks, and then a setRXMode() to switch back into reading mode.  The actual writes don't need to flushed, only when you get back to the setRXMode would you need to flush() and when that completes you would then do your digitalWrite to set chip back to RX mode... 

Again Sorry for maybe jumping the gun.  I will probably wait until I see what your next drop of code works. 

Kurt

OpusK, 2019-11-18:
Hi, @KurtE 

> Sorry, looks like maybe I am jumping the gun to see about writing up a C implementation of SyncRead?

Don't be sorry for me :) I am always grateful for your passionate feedback and learning.

First of all, as I said before, I want to write C only for protocol.
And I want to write master/slave function in CPP.
In other words, anything related to Instruction/Status packets will be written in CPP as it is now.

I need time to review your feedback. And I need time to take a closer look at the SyncRead/Wirte you created with Class.

And, the APIs I mentioned are one of the things I've been thinking about to find a better form as a user API.
Therefore, I am listening to your opinion. Can you suggest me the intuitive API you want as a user, whether it's in the form of your class or another API?
The APIs must be implementable with the functions in protocol.h.

KurtE, 2019-11-18:
Hi @OpusK, 

As I mentioned I was playing with the sketch I had the C++ sync read and write classes in, which are  running fine on current code_layer_redesign code base. 

So I thought I would try implementing the C version, and since I was doing C version, I thought I would start off with your ideas earlier on.   But mainly ran into questions on what your thoughts were on it.   Again like:
```
bool syncRead(uint8_t *p_param, uint16_t param_len, uint8_t id_cnt,
              uint8_t *p_recv_buf, uint16_t recv_buf_capacity, uint32_t timeout_ms = 10);
```
What is expected for each of these arguments: 
Somewhere I need to pass in, which register ID to start with, how many bytes to read, and a list of servo IDs.  So in this case was my guess correct that p_param points to this data that is already formatted into the right arguments for the SYNC_READ message? (2 byte starting)(2 byte length) followed by the IDS?  if so what does id_cnt contain?  Is it simply param_len - 4? 

Things like that. 

Then the next half (or question) was about ok Now if I am going to implement the function syncRead, and I have this data?  How to I get it out to the Serial port...   
In C++, would simpley do something like: 
`  if (!dxl.txInstPacket(DXL_BROADCAST_ID, DXL_INST_SYNC_READ, p_param, param_len)) return false; // failed on tx
`
In my Earlier version I simply used a wrapper function: 
```
extern "C" {
  boolean DXLTxPacketInst(uint8_t id, uint8_t instruction, void* buffer, uint16_t length) {
    return dxl.txInstPacket(id, instruction, (uint8_t*)buffer, length);
  }
Are you thinking of doing something like that as well?   

Likewise a C wrapper to do the RX? 

Some of the other stuff was more a global question on handling big versus little endian.  

As one can not assume doing the logical of: 
uint32_t current_position;
dxl.read(1, CURRENT_POS, 4, &current_position, 4);
```

Will work on different platforms, as the byte ordering may be opposite...

KurtE, 2019-11-19:
@OpusK,

Another quick update on playing with C version.  
First is the ability for a C file to use any of the data structures and constants in the Dynamixl2Arduino header files.  You can not do: 
```
@@ -18,9 +18,13 @@
#define DYNAMIXEL_2_ARDUINO_H_

#ifdef __cplusplus
#include "utility/master.h"
#include "utility/slave.h"
#include "actuator.h"
#else
#include "dxl_c/protocol.h"
#endif

enum OperatingMode{
  OP_POSITION = 0,
@ -48,6 +52,9 @@ enum D2ALibErrorCode
  D2A_LIB_ERROR_UNKNOWN_MODEL_NUMBER
};

// Class only valid in C++
#ifdef __cplusplus

class Dynamixel2Arduino : public DYNAMIXEL::Master
{
  public:
@ -456,5 +463,6 @@ class Dynamixel2Arduino : public DYNAMIXEL::Master

#endif  /* __cplusplus */

#endif /* DYNAMIXEL_2_ARDUINO_H_ */`
```
This is in diff format.  Basically added 2 #ifdef __cplusplus 
with else and so far the only thing that is included in C file is the first defines... 
and it includes the protocol.h...  

Could be done with one #ifdef also could/should also be done, by maybe updating each of the header files includes to only include the C++ stuff when in C++ mode... 

Next back to SyncRead: 
I put in my two wrapper functions to do writes and reads, to call off in C... 
But again wondering about both data going in and data coming out.

**Going in**: const uint8_t *p_param, uint16_t param_len, uint8_t id_cnt
Again: I am assuming that p_param array already has the starting register as the first two bytes and how many bytes to read in the next two bytes?  And then followed by the IDs of the servos to read in.

So what is param_len?  Is it the size of the initialized data in p_param?  or how many bytes per servo?
Likewise id_cnt again I assume how many Servos to do the read on?

Again assuming param_len is the actual size of data passed in, than id_cnt is redundant, as it should simply be param_len-4.

**Received data** 
Again what are the details of p_recv_buf and recv_buf_capacity?  
There are some options on what we may save in here and where.
a) Do we save away the id? 
b) Do we save away the error code (err_idx)?
c) and how do we order things. 

Example: do we save away the data in the order we receive it.  Or does this retrieved data have a specific slot in memory where it should be stored?   That is suppose, we query 10 bytes from 18 servos like my previous example: Suppose we request the data from, the  ID list: 
`const uint8_t ServoIDList[] = {19, 3, 5, 2, 4, 6, 7, 9, 11, 8, 10, 12, 13, 15, 17, 14, 16, 18};`

And suppose suppose one leg stops working and I don't receive data from servos 2, 4, 6.
And assume the data returned is in the order of the servos in the ID list. 
Should the data for ID 7 go into slot 3 or into slot 6? 

If Slot 6, then we don't need to store away the servo ID in the results and helper functions like I mentioned might work off of slot # instead of ID... There may be a helper function to convert ID to slot... 

Again assuming we wish to keep the err_idx for servos.  Do we keep this data spread out in the table, or for example we store them as the first N bytes in the retrieve data. In either case maybe we initialize these slots to some invalid setting, like maybe 0xff to say that servo did not respond... So again we can see which servos did not respond without walking the list... 

Again lots of options :D Not sure which ways make more sense to you?

OpusK, 2019-11-22:
Hi, @KurtE 

Sorry. I was late because of other tasks.

> What is expected for each of these arguments:
Somewhere I need to pass in, which register ID to start with, how many bytes to read, and a list of servo IDs. So in this case was my guess correct that p_param points to this data that is already formatted into the right arguments for the SYNC_READ message? (2 byte starting)(2 byte length) followed by the IDS? if so what does id_cnt contain? Is it simply param_len - 4?

I think that the parameters of the API that I declare arbitrarily can cause user confusion.
Rather, the following functions are more convenient to use.

```cpp
    /* Easy functions for Sync Read */
    bool beginSyncRead(uint16_t addr, uint16_t addr_len);
    bool addSyncReadID(uint8_t id);
    bool sendSyncRead(uint8_t *p_recv_buf, uint16_t recv_buf_capacity);

    /* Easy functions for Sync Write */
    bool beginSyncWrite(uint16_t addr, uint16_t addr_len);
    bool addSyncWriteData(uint8_t id, uint8_t *p_data, uint16_t data_len);
    bool sendSyncWrite();
```

> Are you thinking of doing something like that as well?
Likewise a C wrapper to do the RX?

As I said before, I'm not thinking of making it c except protocol.c.
It would be nice to have a c version for master and slave as well, but for now, I'm thinking of supporting only the CPP version.
However, it seems worth considering the C wrapper you mentioned.

Are you using the Sync feature with the code you created below?
 - [Link](https://github.com/KurtE/Open_CM_CR_Arduino_Sketches/tree/Dynamixel2Arduino/Dynamixel2Arduino_sync_read_write)

KurtE, 2019-11-22:
@OpusK ,

I can imagine you have lots of stuff going on. 

Will take a look at your Easy functions.  Not sure if you are expecting that there will only be one set of Sync reads, and Sync write active at any one time...  

That is do you see in a loop that the user will always do something like: 
```
    beginSyncRead(DXL_X_PRESENT_POSITION, 4);
    for (uint8_t i=0; i < sizeof(ServoIDList); i++) addSyncReadID(ServoIDList[i]);
    if (sendSyncRead(sync_recv_buffer, sizeof(sync_recv_buffer)) {.... }

```
Especially if they have multiple ones...  Or can the do the begin and the adds in setup, and just call sendSyncRead() in loop? 

As for SyncWrite, still an issue of memory.  Do you see that the API will malloc or the like buffers as needed?  If so, potentially could add another parameter to both beginMethods, like maybe, something like max number of IDS that it will support?  Maybe also for those who do not want any heap stuff, maybe add method, like to set buffers?  Something like:

```
    bool beginSyncRead(uint16_t addr, uint16_t addr_len, uint8_t cnt_ids);
    bool addSyncReadID(uint8_t id);
    bool sendSyncRead(uint8_t *p_recv_buf, uint16_t recv_buf_capacity);
    bool setSyncReadBuffer(uint8_t *buffer, uint16_t size); // should be >= 4 + cnt_ids

    /* Easy functions for Sync Write */
    bool beginSyncWrite(uint16_t addr, uint16_t addr_len, cnt_ids);
    bool addSyncWriteData(uint8_t id, uint8_t *p_data, uint16_t data_len);
    bool sendSyncWrite();
    bool setSyncWriteBuffer(uint8_t *buf, uint16_t size);
```
Will try changing to some of this over the weekend.  

Yes that is the link to where several test apps have been.   I have not pushed up the most recent of this conversation as it does not compile...  But hopefully by Monday...

OpusK, 2019-11-25:
@KurtE 

> bool setSyncReadBuffer(uint8_t *buffer, uint16_t size); // should be >= 4 + cnt_ids

I was considering this function as well.
I thought I could create and use different Sync packets in different buffers.

Of course, I'm wondering if it's better to add this to the Master class or use a separate class as you created.

KurtE, 2019-11-25:
@OpusK,

Yesterday I did hack in a version of syncRead along the line of your APIs as part of the one test sketch and update the main sketch to call it.  

It builds, but I have not tested it yet... But I did push it up the github project in case you wanted a look. 

So far I have not put in any of the set Buffers, but did add some query functions to extract the data retrieved from the buffer.  Currently I have the buffer encoded with data sort of like:
```
Byte 0 - how many IDS were in list, 
Byte 1 - How many were retrieved
Bytes 2-3 Data length for items...(N)
Followed by data: 
4 - ID element 1
5 - Err
6-(6+N-1) data for element

6+N - ID element 2
```

Will hopefully test later today. 
...

OpusK, 2019-11-26:
@KurtE 

I created and used an API to change the buffer as shown below, but in the end, it seemed better to manage it as a separate class to support querying on received data. (Changing the buffer, there was no problem communicating.)

```cpp
typedef struct InfoSyncBulkInst{
  uint8_t* p_packet_buf;
  uint16_t packet_buf_capacity;
  uint16_t packet_length;
  uint8_t id_cnt;
  bool is_complete_packet;
  uint16_t addr_len;
} InfoSyncBulkInst_t;

bool beginSyncRead(uint16_t addr, uint16_t addr_len, InfoSyncBulkInst_t *p_sync_info = nullptr);
bool addSyncReadID(uint8_t id, InfoSyncBulkInst_t *p_sync_info = nullptr, bool flag_end_add = false);
uint16_t sendSyncRead(uint8_t *p_recv_buf, uint16_t recv_buf_capacity, InfoSyncBulkInst_t *p_sync_info = nullptr);

bool beginSyncWrite(uint16_t addr, uint16_t addr_len, InfoSyncBulkInst_t *p_sync_info = nullptr);
bool addSyncWriteData(uint8_t id, uint8_t *p_data, InfoSyncBulkInst_t *p_sync_info = nullptr, bool flag_end_add = false);
bool sendSyncWrite(InfoSyncBulkInst_t *p_sync_info = nullptr);
```

Below is a simple example.
Of course, I'm still looking for a better way.

```cpp
#include <Dynamixel2Arduino.h>

// Please modify it to suit your hardware.
#if defined(ARDUINO_AVR_UNO) || defined(ARDUINO_AVR_MEGA2560) // When using DynamixelShield
  #include <SoftwareSerial.h>
  SoftwareSerial soft_serial(7, 8); // DYNAMIXELShield UART RX/TX
  #define DXL_SERIAL   Serial
  #define DEBUG_SERIAL soft_serial
  const uint8_t DXL_DIR_PIN = 2; // DYNAMIXEL Shield DIR PIN
#elif ARDUINO_OpenCM904 // Official ROBOTIS board with DXL circuit.
  #define DXL_SERIAL   Serial3 //OpenCM9.04 EXP Board's DXL port Serial. (Serial1 for the DXL port on the OpenCM 9.04 board)
  #define DEBUG_SERIAL Serial
  const uint8_t DXL_DIR_PIN = 22; //OpenCM9.04 EXP Board's DIR PIN. (28 for the DXL port on the OpenCM 9.04 board)
#elif ARDUINO_OpenCR // Official ROBOTIS board with DXL circuit.
  // For OpenCR, there is a DXL Power Enable pin, so you must initialize and control it.
  // Reference link : https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/DynamixelSDK/src/dynamixel_sdk/port_handler_arduino.cpp#L78
  #define DXL_SERIAL   Serial3
  #define DEBUG_SERIAL Serial
  const uint8_t DXL_DIR_PIN = 84; // OpenCR Board's DIR PIN.    
#else // When using DynamixelShield
  #define DXL_SERIAL   Serial1
  #define DEBUG_SERIAL Serial
  const uint8_t DXL_DIR_PIN = 2; // DYNAMIXEL Shield DIR PIN
#endif

/* InfoSyncBulkInst_t
  A structure that contains the information needed for the parameters of the 'Sync/Bulk Read/Write packet'.

typedef struct InfoSyncBulkInst{
  uint8_t* p_packet_buf;
  uint16_t packet_buf_capacity;
  uint16_t packet_length;
  uint8_t id_cnt;
  bool is_complete_packet;
} InfoSyncBulkInst_t;
*/

DYNAMIXEL::InfoSyncBulkInst_t sr_present_pos;
const uint16_t user_pkt_buf_cap = 128;
uint8_t user_pkt_buf[user_pkt_buf_cap];
const uint16_t recv_buf_cap = 128;
uint8_t recv_buf[recv_buf_cap];
int32_t goal_vel = 0;

Dynamixel2Arduino dxl(DXL_SERIAL, DXL_DIR_PIN);

void setup() {
  // put your setup code here, to run once:
  DEBUG_SERIAL.begin(115200);
  dxl.begin(1000000);
  dxl.scan();
  
  dxl.torqueOff(1);
  dxl.setOperatingMode(1, OP_VELOCITY);
  dxl.torqueOn(1);

  dxl.torqueOff(3);
  dxl.setOperatingMode(3, OP_VELOCITY);
  dxl.torqueOn(3);

  // Using user buffer
  sr_present_pos.p_packet_buf = user_pkt_buf;
  sr_present_pos.packet_buf_capacity = user_pkt_buf_cap;

  dxl.beginSyncRead(132, 4, &sr_present_pos);
  dxl.addSyncReadID(1, &sr_present_pos);
  dxl.addSyncReadID(3, &sr_present_pos, true);

  // Using class default buffer
  dxl.beginSyncWrite(104, 4);
  goal_vel = 100;
  dxl.addSyncWriteData(1, (uint8_t*)&goal_vel);
  goal_vel = 200;
  dxl.addSyncWriteData(3, (uint8_t*)&goal_vel);
  dxl.sendSyncWrite();
}

void loop() {
  // put your main code here, to run repeatedly:
  uint16_t recv_len;
  
  recv_len = dxl.sendSyncRead(recv_buf, recv_buf_cap, &sr_present_pos);
  if(recv_len > 0){
    DEBUG_SERIAL.print("\t Receive Len : ");
    DEBUG_SERIAL.print(recv_len);
    DEBUG_SERIAL.println();
    for(uint16_t i=0; i<recv_len; i++){
      DEBUG_SERIAL.print(recv_buf[i], HEX);DEBUG_SERIAL.print(" ");
    }
    DEBUG_SERIAL.println();
  }else{
    DEBUG_SERIAL.print("\t Fail!!   ");
    DEBUG_SERIAL.print(dxl.getLastLibErrCode());
    DEBUG_SERIAL.println();
  }
  delay(500);
}
```
https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/code_layer_redesign/examples/advanced/sync_read_write_raw/sync_read_write_raw.ino

KurtE, 2019-11-28:
Hi Again,  Will take a look at the above. 

I just had a chance to get my earlier sketch to work, with a C version of SyncRead.   Pushed up current stuff, will take a look in the next day or two at your newer stuff. 

Kurt

KurtE, 2019-11-29:
@OpusK ,

I synced up to your branch again and took a look at your example, which looks like you are getting there. 

A few comments.

Not sure if better on addSyncReadID to have the 3rd parameter or simply have another method, like 
endSetupSyncRead()... - probably can find better name.   But what you have works.

in the method: sendSyncRead(), you are assuming that all of the servos in the list will respond and in the order in which they are in the list.  But what happens if one of the servos does not respond? 
i.e suppose you ask for servos 1, 2, 3, 4, 5, 6
and servo ids 3 and 5 does not respond?  You loop 6 types waiting for message:
  ```
for(uint8_t i=0; i<p_info->id_cnt; i++)
  {
    if(rxStatusPacket(&p_recv_buf[i*p_info->addr_len], 
    recv_buf_capacity-i*p_info->addr_len, 3) != nullptr){
      recv_cnt++;
    }
  }
```
The data in your buffers you passed in , will probably write in the first 4 elements of it, for servos: 1, 2, 4, 6

And your test app will assume that for example index 2 is for Servo #3, and there is no way to verify which one it is actually for? As the loop above does not check IDS.   There are at least a few options here?

a) Save the IDs as part of the user array... So they can loop through and know which ID they are looking at...

b) Have the loop code above verify the ID, matches, the expected one, if not, move the data down to where it should be in the array  to the index expected, and probably clear out the data for the one missed?  Some error value? 

Also again not sure if you wish to worry about, but you are also not keeping the ERR values from the received servos, so for example if a servo has hardware issue, like it detected a low voltage, you will not get that information through the Sync Read. 

Also it will be interesting to see if all of the boards you wish to support all use little endian? If not examples like the one that assumes that the 10 bytes are stored in the same format as what your servos expect and/or receive could mess up.

KurtE, 2019-12-01:
Quick message/suggestion: 

In the SerialPortHandler
```
class SerialPortHandler : public DXLPortHandler
{
  public:
    SerialPortHandler(HardwareSerial& port, const int dir_pin = -1);

    virtual void begin() override;
    virtual void end() override;
    virtual int available(void) override;
    virtual int read() override;
    virtual size_t write(uint8_t) override;
    virtual size_t write(uint8_t *buf, size_t len) override;

    void begin(unsigned long baud);
    unsigned long getBaud() const;

  private:
    HardwareSerial& port_;
    const int dir_pin_;
    unsigned long baud_;
};

```
I think it might be beneficial if your new begin(baud) method was virtual.
That is: 
`   virtual void begin(unsigned long baud);`
As for example suppose I am using this for a Teensy (or other hardware that supports hardware half duplex).  And as part of the SerialX.begin, I may need to then do other calls or muck with registers, in order to initialize that serial port into half duplex mode.  

Sort of for example how you have code added in the default specific to OpenCM or OpenCR...

OpusK, 2019-12-03:
Hi, @KurtE 

I'm sorry for late reply.
First of all, I only applied changing the virtual function.

Can I leave an answer for the sync function later?

KurtE, 2019-12-03:
@OpusK 

Thanks, sounds good.  

As for Sync functions and the like, no worry, no hurry...

OpusK, 2019-12-13:
Hi, @KurtE 

> Not sure if better on addSyncReadID to have the 3rd parameter or simply have another method, like
endSetupSyncRead()... - probably can find better name. But what you have works.

I also worried about this a lot. As you comment I may need to modify the names of the functions somewhat.
```cpp
beginSetupSyncRead
addSyncReadID
endSetupSyncRead
sendSyncRead
```

> in the method: sendSyncRead(), you are assuming that all of the servos in the list will respond and in the order in which they are in the list. But what happens if one of the servos does not respond?

Yes. This is my mistake
If a protocol does not receive a response from one ID, it will not receive a response from the rest of the IDs, so a break statement is added to prevent unnecessary work.

```cpp
  for(uint8_t i=0; i<p_info->id_cnt; i++)
  {
    if(rxStatusPacket(&p_recv_buf[i*p_info->addr_len], 
    recv_buf_capacity-i*p_info->addr_len, 3) != nullptr){
      recv_cnt++;
    }else{
      break;
    }
  }
```

> And your test app will assume that for example index 2 is for Servo #3, and there is no way to verify which one it is actually for? As the loop above does not check IDS. There are at least a few options here?

In fact, this example assumes that the user enters the ID correctly.
Because it's a raw API, I've planned to do that in a separate class like yours.

By the way, when I try to create such a class, do I need to create these raw APIs?
If I think about it now, it doesn't seem necessary.
It would be better to create a separate class like yours with the functions below.
```cpp
    // raw APIs
    bool txInstPacket(uint8_t id, uint8_t inst_idx, uint8_t *p_param, uint16_t param_len);
    const InfoToParseDXLPacket_t* rxStatusPacket(uint8_t* p_param_buf, uint16_t param_buf_cap, uint32_t timeout_ms = 10);
```

> Also again not sure if you wish to worry about, but you are also not keeping the ERR values from the received servos, so for example if a servo has hardware issue, like it detected a low voltage, you will not get that information through the Sync Read.

The same seems to be the case.
Rather, it seems that the form of the function I created earlier met these parts. (There was a waste of memory.)
Perhaps this is better if we manage it in a Class form similar to yours.

> Also it will be interesting to see if all of the boards you wish to support all use little endian? If not examples like the one that assumes that the 10 bytes are stored in the same format as what your servos expect and/or receive could mess up.

Currently, I am writing only a little endian. However, I think it's a good idea to allow endian users to choose and use it in the future.
Endianness is about buffering, so we tried to consider it after the rest of the functionality had stabilized.
Is there any code structure you think about this?

OpusK, 2019-12-26:
Hi, @KurtE 

I gave up several levels of rawAPI and changed back to the previous structure parameterized API. (Struct member changed)

Based on this, I plan to create a separate class and provide it like a class you create or a class method in the SDK and the workbench.

OpusK, 2019-12-26:
@KurtE,

I simply created template classes. What do you think about this?

- [class definitions in master.h](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/59c93046802ceafc2a5c9c5986d85fd322022da7/src/utility/master.h#L195)
- [sync_read_write.ino](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/59c93046802ceafc2a5c9c5986d85fd322022da7/examples/advanced/sync_read_write/sync_read_write.ino)

[4 more replies]

closed 2020-09-21 by OpusK