ROBOTIS-GIT/Dynamixel2Arduino #26: Broadcast ID does not work correctly.

asdfire1, 2019-11-12:
The reboot function does not work when you use broadcast id.
Those lines in the reboot function in master.cpp prevent it
`if (id == DXL_BROADCAST_ID){
   last_lib_err_code_ = DXL_LIB_ERROR_NOT_SUPPORT_BROADCAST;
   return false;
  }`
When they are commented out the reboot works fine for all connected dynamixels. Why is it in the code?

In the same function the reboot code does not work at all if this is left as is
`last_lib_err_code_ = dxlTxPacketInst(&packet_, id, INST_REBOOT, tx_param, 1);`
once it is changed to 
`last_lib_err_code_ = dxlTxPacketInst(&packet_, id, INST_REBOOT, tx_param, 0);`
the reboot starts working.
When you try to set the torqueOn to broadcast ID it does not work either.

OpusK, 2019-11-13:
Hi, @asdfire1 

Thanks for your feedback!

We're currently working on a number of [updates](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/issues/23), including the API.
Previously, [the param code for reboot was also modified](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/code_layer_redesign/src/utility/master.cpp#L466).
I will modify the broadcast-related to reboot.

OpusK, 2020-02-07:
Hi, @asdfire1 

I have released a modified version of this issue.
Please refer to [0.4.0 version](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/releases/tag/0.4.0).

closed 2020-03-24 by OpusK