ROBOTIS-GIT/Dynamixel2Arduino #31: Can not scan the Dynamixel 2xl430-w250-t

yuan0623, 2020-04-08:
Hi
I recently purchased one `2xl430-w250-t` and one `Arduino shield`. My goal is to use Arduino to control the Dynamixel motor. I have installed the `Dynamixel2Arduino` library through library manager. I have mounted the Arduino shield on top of `Arduino Mega 2560`. The Arduino Mega 2560 is connected to my Mac through a cable (USB-B to USB-C), the TTL cable is connecting the Arduino shield and one port of the `Dynamixel 2xl430-w250-t` (ID 1). I see the LED blink once when I turn on the switch of the `Arduino shield`.  After I have uploaded the code, I switch the UART switch. The picture of my setup can be checked [here](https://photos.app.goo.gl/bfJz5Dowh5WcXXqx9).
However, I have encountered the problem:
1. When I run the scan Dynamixel, the system tell me they found 0 motor. (I have to choose Arduino Mega ADK to have a proper print), the screenshot can be found [here](https://photos.app.goo.gl/jH9tEEf8aaued9AS7).
2. When I do same thing but choosing Mega 2560, the print message is a bunch of unreadable “?” .
3. I try to use Arduino UNO to do the same thing, the print message is a bunch of unreadable “?” 
The screenshot of problem 2 and 3 can be found [here](https://photos.app.goo.gl/bT3ZVvmpSJW3CEqT8).

I have searched the forum, but I didn’t find anything useful. Can you give me some advice on this?

Best

Yuan

OpusK, 2020-04-08:
Hi, @yuan0623 

Looking at your first picture, the switch is still on the UART.
At run time, you should switch to DYNAMIXEL.
In this regard, please check the manual link below.
 - [DYNAMIXEL Shield's Switch](http://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#switches)

This is also the reason why the serial monitor is broken.
Also, due to the hardware structure of UNO and Mega, when using a serial monitor, please refer to the link below.

- [Use of Serial Monitor with DYNAMIXEL Shield](http://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#use-of-serial-monitor-with-dynamixel-shield)

yuan0623, 2020-04-09:
> Hi, @yuan0623
>
> Looking at your first picture, the switch is still on the UART.
> At run time, you should switch to DYNAMIXEL.
> In this regard, please check the manual link below.
>
> * [DYNAMIXEL Shield's [...]

Hi, thank you so much for your reply!
I actually switch the UART to the Dynamixel as well.  However, the problem is still there, nothing has changed at all. I still have the exact same problem. The picture can be seen [here](https://photos.app.goo.gl/NJZDrV5yngRdwDnU8).

I was wondering, do I need to modify the sample code? It seems that the sample code should work on Arduino Uno and Mega without modification, right?

OpusK, 2020-04-10:
> I was wondering, do I need to modify the sample code? It seems that the sample code should work on Arduino Uno and Mega without modification, right?

- [Use of Serial Monitor with DYNAMIXEL Shield](http://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#use-of-serial-monitor-with-dynamixel-shield)

Have you checked the contents of the link above?
Unfortunately, this is a well-known issue and method by MEGA and UNO sharing USB port 0,1.
So you have to use the method described in the manual above or some other method I will present here.

To use without modifying the sample code, you must use Serial Monitor in a different way than USB.
The method is SoftwareSerial.
But, this method requires a converter module such as `UART to USB`. 

As you know, MEGA2560 has 4 HardwareSerial.
Therefore, if you do not have a `UART to USB` device, I recommend that you make some modifications to the code and some hardware like below.

1. Make pins 0 and 1 of the shield L-shaped. (To avoid connecting with pins 0 and 1 of Mega)
[image]

2. Connect another HardwareSerial pin of Mega and 0(RX),1(TX) pin of Shield with a jumper.
[image]

3. Modify the code because the serial number has changed like below. (Original to New)
    The two parts that need to be modified are DEBUG_SERIAL and DXL_SERIAL.

- Original
```cpp
// Please modify it to suit your hardware.
#if defined(ARDUINO_AVR_UNO) || defined(ARDUINO_AVR_MEGA2560) // When using DynamixelShield
  #include <SoftwareSerial.h>
  SoftwareSerial soft_serial(7, 8); // DYNAMIXELShield UART RX/TX
  #define DXL_SERIAL   Serial3
  #define DEBUG_SERIAL Serial
  const uint8_t DXL_DIR_PIN = 2; // DYNAMIXEL Shield DIR PIN
#elif defined(ARDUINO_SAM_DUE) // When using DynamixelShield
  ...
  ...
```

- New
```cpp
// Please modify it to suit your hardware.
#if defined(ARDUINO_AVR_UNO)
  #include <SoftwareSerial.h>
  SoftwareSerial soft_serial(7, 8); // DYNAMIXELShield UART RX/TX
  #define DXL_SERIAL   Serial3
  #define DEBUG_SERIAL Serial
  const uint8_t DXL_DIR_PIN = 2; // DYNAMIXEL Shield DIR PIN
#elif defined(ARDUINO_AVR_MEGA2560) // When using DynamixelShield
  #define DXL_SERIAL   Serial3
  #define DEBUG_SERIAL Serial
  const uint8_t DXL_DIR_PIN = 2; // DYNAMIXEL Shield DIR PIN
#elif defined(ARDUINO_SAM_DUE) // When using DynamixelShield
  ...
  ...
```

yuan0623, 2020-04-14:
thanks for your perfect explanation. I tried your way and now I have scanned all my motors!

closed 2020-04-14 by yuan0623

itsameWolf, 2021-09-22:
Hello, i've been having a similar problem. I'm trying to use an arduino mega to control 4 AX12A. I followed the instruction for the modifications to the shield and to the code but the Arduino IDE throws the follwing error
`C:\Users\LManes\AppData\Local\Temp\ccoJUeWl.ltrans0.ltrans.o: In function `end':
C:\Users\LManes\Documents\Arduino\libraries\Dynamixel2Arduino\src\utility/port_handler.cpp:65: undefined reference to `HardwareSerial::end()'
C:\Users\LManes\AppData\Local\Temp\ccoJUeWl.ltrans0.ltrans.o: In function `begin':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:121: undefined reference to `HardwareSerial::begin(unsigned long, unsigned char)'
C:\Users\LManes\AppData\Local\Temp\ccoJUeWl.ltrans0.ltrans.o: In function `__base_ctor ':
C:\Users\LManes\Documents\Arduino\libraries\Dynamixel2Arduino\src\utility/master.cpp:35: undefined reference to `operator new[](unsigned int)'
C:\Users\LManes\AppData\Local\Temp\ccoJUeWl.ltrans0.ltrans.o: In function `__base_ctor ':
C:\Users\LManes\Documents\Arduino\libraries\Dynamixel2Arduino\src/Dynamixel2Arduino.cpp:122: undefined reference to `operator new(unsigned int)'
C:\Users\LManes\AppData\Local\Temp\ccoJUeWl.ltrans0.ltrans.o: In function `__base_ctor ':
C:\Users\LManes\Documents\Arduino\libraries\Dynamixel2Arduino\src\utility/port_handler.cpp:24: undefined reference to `Serial3'
C:\Users\LManes\Documents\Arduino\libraries\Dynamixel2Arduino\src\utility/port_handler.cpp:24: undefined reference to `Serial3'
C:\Users\LManes\AppData\Local\Temp\ccoJUeWl.ltrans0.ltrans.o:(.rodata+0x1a): undefined reference to `Print::write(unsigned char const*, unsigned int)'
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avr6/crtatmega2560.o:(.init9+0x0): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
`
I understand it is something related to the port handler part of the library, but I don't know how to fix the error