ROBOTIS-GIT/Dynamixel2Arduino #41: Get factory limits / defaults from Servo

LegacyMecha, 2020-08-13:
I need to get the range of numbers possible for the Min and Max limits but the function "getModelDependencyFuncInfo" is private and not sure how to use it. I would love to have something like "readControlTableItemDefaultValue(MAX_POSITION_LIMIT, DXL_ID)" to find out what the factory ranges are on a particular setting, without changing the settings. I can then use this to make some calculations.

ROBOTIS-Will, 2020-08-13:
Hi,
Min / Max limits and default values are not saved in the DYNAMIXEL Control Table so you cannot access them, but you can get those information from the eManual of each product.
https://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#control-table-of-eeprom-area

LegacyMecha, 2020-08-14:
Not true, the library has all the info. The class checks to see if any input is within range of a specific model so the servo doesn't get the wrong values. These and other functions do this. I'm just asking to make a public version of this information so I don't have to modify the library.

`  item_info = getControlTableItemInfo(model_num, item_idx);`

```
typedef struct ModelDependencyFuncItemAndRangeInfo{
  uint8_t func_idx; //enum Functions
  uint8_t item_idx; //enum ControlTableItem
  uint8_t unit_type; //enum ParamUnit
  int32_t min_value;
  int32_t max_value;
  float unit_value;
} ModelDependencyFuncItemAndRangeInfo_t;

typedef struct ItemAndRangeInfo{
  uint8_t item_idx; //enum ControlTableItem
  uint8_t unit_type; //enum ParamUnit
  int32_t min_value;
  int32_t max_value;
  float unit_value;
} ItemAndRangeInfo_t;

static ItemAndRangeInfo_t getModelDependencyFuncInfo(uint16_t model_num, uint8_t func_num);
static float f_map(float x, float in_min, float in_max, float out_min, float out_max);
static bool checkAndconvertWriteData(float in_data, int32_t &out_data, uint8_t unit, ItemAndRangeInfo_t &item_info);
static bool checkAndconvertReadData(int32_t in_data, float &out_data, uint8_t unit, ItemAndRangeInfo_t &item_info);

```

Here's some values in the library, but how can I access them:

```
/* PRO R series */
const ModelDependencyFuncItemAndRangeInfo_t dependency_pro_r_m42_10[] PROGMEM = {
#if (ENABLE_ACTUATOR_PRO_R)
  {SET_POSITION, GOAL_POSITION, UNIT_DEGREE, -2147483647, 2147483647, 0.00136785},
  {GET_POSITION, PRESENT_POSITION, UNIT_DEGREE, -2147483647 , 2147483647, 0.00136785},

  {SET_VELOCITY, GOAL_VELOCITY, UNIT_RPM, -8000, 8000, 0.00389076},
  {GET_VELOCITY, PRESENT_VELOCITY, UNIT_RPM, -8000, 8000, 0.00389076},

  {SET_CURRENT, GOAL_TORQUE, UNIT_MILLI_AMPERE, -900, 900, 4.02832},
  {GET_CURRENT, PRESENT_CURRENT, UNIT_MILLI_AMPERE, -900, 900, 4.02832},
#endif
  {LAST_DUMMY_FUNC, ControlTableItem::LAST_DUMMY_ITEM, UNIT_RAW, 0, 0, 0}
};
```

ROBOTIS-Will, 2020-08-14:
You're right, some of the min / max values are defined in the library, but those are very limited usage for unit calculation. 
For example, min / max position values are only the min / max number that can be represented with 32bit integer and do not represent the actual single turn min/max position values of DYNAMIXEL which could vary by product.

LegacyMecha, 2020-08-14:
I need the Arduino to restore the default MIN/MAX travel for ANY random servo model. How can I do that? I CAN NOT use a computer to do this in my situation.

ROBOTIS-Will, 2020-08-17:
Since our Arduino libraries should be able to run and compile on memory limited Arduino boards such as Uno, it doesn't come with various information. You can modify the source code and add Min/Max values of each model so that they can be acquired in your source code.
Thank you.

LegacyMecha, 2020-08-17:
So what happens if I send a write command to a dynamixel that's out of it's range? like write MIN_LIMIT_POSITION to "-1" on a 0-4095 range model?

LegacyMecha, 2020-08-24:
For example, how can I read this info into my code so i can use it? It's already in flash memory when it's compiled, so how can I read it?

```
const ModelDependencyFuncItemAndRangeInfo_t dependency_pro_r_m54_40[] PROGMEM = {
#if (ENABLE_ACTUATOR_PRO_R)
  {SET_POSITION, GOAL_POSITION, UNIT_DEGREE, -2147483647, 2147483647, 0.00143188},
  {GET_POSITION, PRESENT_POSITION, UNIT_DEGREE, -2147483647 , 2147483647, 0.00143188},

  {SET_VELOCITY, GOAL_VELOCITY, UNIT_RPM, -8000, 8000, 0.00397746},
  {GET_VELOCITY, PRESENT_VELOCITY, UNIT_RPM, -8000, 8000, 0.00397746},  

  {SET_CURRENT, GOAL_TORQUE, UNIT_MILLI_AMPERE, -360, 360, 16.11328},
  {GET_CURRENT, PRESENT_CURRENT, UNIT_MILLI_AMPERE, -360, 360, 16.11328},
#endif
  {LAST_DUMMY_FUNC, ControlTableItem::LAST_DUMMY_ITEM, UNIT_RAW, 0, 0, 0}
};
```

ROBOTIS-Will, 2020-08-25:
> So what happens if I send a write command to a dynamixel that's out of it's range? like write MIN_LIMIT_POSITION to "-1" on a 0-4095 range model?

Writing invalid value to DYNAMIXEL will cause a Data Range Error, but the current DYNAMIXEL2Arduino library does not use this [error status](https://emanual.robotis.com/docs/en/dxl/protocol2/#error).

In order to access those data in the code, you can create custom public functions that reads from the struct.
Thank you.

LegacyMecha, 2020-08-25:
Can you give me a simple example given I have a model number how to get the "UNIT_DEGREE" value from it?

`
model_number = dxl.getModelNumber(DXL_ID); // Like "1010" for "XH430-W210-R" 
degree_divider = (get UNIT_DEGREE HERE) ???

`

ROBOTIS-Will, 2020-09-21:
Sorry about the late reply, but we do not provide a sample code for a specific application as there are too many similar requests.
Thank you for your understanding.

ROBOTIS-Will, 2020-12-21:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-21 by ROBOTIS-Will