ROBOTIS-GIT/Dynamixel2Arduino #46: slave: handling of 'ROM' control items

sonelu, 2020-10-05:
At this moment the slave class does not include any retrieval / save of control items to ROM (ex. ID, protocol, baud-rate, etc.). If devices are really supposed to be build with this library we would need support for saving these configuration items into ROM (like they happen for Dynamixel servos). Is that planned to be added at any time?

ROBOTIS-Will, 2020-10-06:
Hi @sonelu,
Thank you for your inquiry.
The slave class was intended to use DYNAMIXEL protocol on an embedded board, and does not targeting to burn into ROM.
As of now, further development on this class is not planned.

sonelu, 2020-10-07:
Would you consider pull requests?

At the moment this repo is the most convenient framework for developing custom Dynamixel devices (ex. sensors) and it would be a shame not to encourage the community to use it.

But for it to be usable it would need to allow some standard functionality for devices (like saving the 'ROM' registers between power cycles). There might be a few other welcome additions, like support for more instructions (sync and bulk read / write, reboot, reset, etc.).

ROBOTIS-Will, 2020-10-08:
Hi @sonelu ,
Yes we are open to collaboration as this is an open source repository.
As long as the request is intended for a general purpose, we will consider to accept the changes.
FYI, you may want to refer some similar feature we developed for other project.
https://github.com/ROBOTIS-GIT/XELNetworkMaster/blob/master/examples/CommXEL-W/CommXEL-W.ino#L372

However, appending additional slave features are currently restricted by our internal regulation.
Thank you.

sonelu, 2020-10-08:
Thanks @ROBOTIS-Will for the link too the CommXEL. That was what I was looking for.

closed 2020-10-08 by sonelu