ROBOTIS-GIT/Dynamixel2Arduino #47: MX-28 and MX64 library compatibility

eliro2, 2020-10-07:
I have tested 2 MX-28's and an MX-64 with the provided library and have been having issues with getting examples to work. The LED example works for all three servos, meaning the id, servo baud_rate, and protocol (1 for all) are all correct. I also verified these values through the dynamixel wizard, and they all correspond to the defaults. For one of the 28's, velocity_mode works as well, which is promising. However, the most important example I would like to get working is position_mode, and that doesn't work for any of them. Any feedback would be appreciated, thanks.

My control environment is a Dynamixel Shield, Arduino UNO, the Dynamixel2Arduino library (on one laptop included through arduino's ide and on another manually added), the MX-28/64 servo models, and a SMPS 12V 5A for power connected to the arduino itself.

ROBOTIS-Will, 2020-10-08:
Hi @eliro2 ,
Thank you for your inquiry.
Have you installed DYNAMIXELShield library?
If you are using a DYNAMIXEL Shield on top of Arduino boards, you should use DYNAMIXEL Shield library examples instead of DYNAMIXEL2Arduino's.
Thank you.

eliro2, 2020-10-08:
I will try this tomorrow, thank you for letting me know! I will reply if it
works or if other issues still persist.

ROBOTIS-Will, 2020-12-21:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-21 by ROBOTIS-Will