ROBOTIS-GIT/Dynamixel2Arduino #49: XM540 Motors are not moving/responding

Sal-Hussain, 2020-10-08:
Hi,
I'm was using the outdated OpenCM IDE. However, I decided to transfer to Arduino I realized that my old code does not work anymore. I tried looking at Dyanmixel2Arduino and use the documentation to convert the code. However, it appears my motors are not responding. I tried running the example code and even then I am getting no response from the motors. I was wondering if anyone can help me with this? What library was the old OpenCM IDE using? I can't tell because I don't have any include statements. It uses Dynamixel.h but the Arduino version does not include methods such as writeDword and readByte.  
[Arduino Files.zip](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/files/5351551/Arduino.Files.zip)

ROBOTIS-Will, 2020-10-12:
Hi @Sal-Hussain 
If you are using OpenCM9.04 + OpenCM485EXP to control the XM540, you can keep using the Arduino IDE with `DYNAMIXEL2Arduino` library. However, if you are using Arduino boards with DYNAMIXEL Shield, you should use `DYNAMIXEL Shield` library. Please refer to [eManual](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#dynamixel-shield-libraries) for more information.

Unfortunately, OpenCM IDE code is not compatible with Arduino IDE and you should modify your code based on the provided example in the library.
For example, `writeDword` function does not exist anymore in the Arduino library so you should use [frequently used functions](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#dynamixel2arduino-class) `setGoalVelocity` or `writeControlTableItem` for writing a specific control table item.

Thank you.

ROBOTIS-Will, 2020-12-21:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-21 by ROBOTIS-Will