ROBOTIS-GIT/Dynamixel2Arduino #52: Multiple Control Dynamixel XL430 XL320

WilliamH07, 2020-10-22:
Hello everyone,
I'm starting to code a code with Dynamixel2Arduino library to control multiple xl430 and x320 (2x xl430 2x x320). 
I use : position_mode to control my dynamixels, it works fine with only 1 dynamixel but I want to control 4 dynamixels in all. To control another dynamxiel I duplicate the code (I post the code I modified below). But with this only 1 dynamixel works (DXL_1). I don't understand why, can someone please help me?
Thank you,
WilliamH07

```

const uint8_t DXL_ID = 1;
const uint8_t DXL_ID2 = 10;
const float DXL_PROTOCOL_VERSION = 2.0;

Dynamixel2Arduino dxl(DXL_SERIAL, DXL_DIR_PIN);

//This namespace is required to use Control table item names
using namespace ControlTableItem;

void setup() {
  // put your setup code here, to run once:
  
  // Use UART port of DYNAMIXEL Shield to debug.
  DEBUG_SERIAL.begin(115200);

  // Set Port baudrate to 57600bps. This has to match with DYNAMIXEL baudrate.
  dxl.begin(1000000);
  // Set Port Protocol Version. This has to match with DYNAMIXEL protocol version.
  dxl.setPortProtocolVersion(DXL_PROTOCOL_VERSION);
  // Get DYNAMIXEL information
  dxl.ping(DXL_ID);
  dxl.ping(DXL_ID2);
  // Turn off torque when configuring items in EEPROM area
  dxl.torqueOff(DXL_ID);
  dxl.torqueOff(DXL_ID2);
  dxl.setOperatingMode(DXL_ID, OP_POSITION);
  dxl.setOperatingMode(DXL_ID2, OP_POSITION);
  dxl.torqueOn(DXL_ID);
  dxl.torqueOn(DXL_ID2);
  
}

void loop() {
  // put your main code here, to run repeatedly:
  
  // Please refer to e-Manual(http://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/) for available range of value. 
  // Set Goal Position in RAW value
  dxl.setGoalPosition(DXL_ID, 512);
  dxl.setGoalPosition(DXL_ID2, 512);
  delay(1000);
  // Print present position in raw value
  DEBUG_SERIAL.print("Present Position(raw) : ");
  DEBUG_SERIAL.println(dxl.getPresentPosition(DXL_ID));
  DEBUG_SERIAL.print("Present Position(raw)Dynamixel2 : ");
  DEBUG_SERIAL.println(dxl.getPresentPosition(DXL_ID2));
  delay(1000);

  // Set Goal Position in DEGREE value
  dxl.setGoalPosition(DXL_ID, 5.7, UNIT_DEGREE);
  dxl.setGoalPosition(DXL_ID2, 5.7, UNIT_DEGREE);
  delay(1000);
  // Print present position in degree value
  DEBUG_SERIAL.print("Present Position(degree) : ");
  DEBUG_SERIAL.println(dxl.getPresentPosition(DXL_ID, UNIT_DEGREE));
  DEBUG_SERIAL.print("Present Position(degree)Dynamxiel2 : ");
  DEBUG_SERIAL.println(dxl.getPresentPosition(DXL_ID2, UNIT_DEGREE));
  delay(1000);
}
```

ROBOTIS-Will, 2020-10-23:
Hi @WilliamH07 
Have you set the 2nd DYNAMIXEL ID to `10` as shown in your code?
DYNAMIXEL works with digital packets and each DYNAMIXEL should be properly assigned with non-duplicated ID.
Also, the default baudrate of XL430 is 57600bps while XL-320 is 1Mbps, so you need to use an identical baudrate for using these two DYNAMIXEL at the same time.
And the most important thing is the power. XL430 runs off of 11.1V while XL-320 runs with 7.4V.
In order to configure the DYNAMIXEL, I recommend to use [DYNAMIXEL Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/) with U2D2.
Thanks.

WilliamH07, 2020-10-23:
Hello @ROBOTIS-Will thanks for the reply,
I use the **U2D2**  with [DYNAMIXEL Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/)  in order to configure the Dynamixel, I set all dynamixels at the same time Baud rate (`1 Mbps`).
For the first  **XL430** I put the id  `10 `and for the second  **XL430**  I put the id `11`.
And for the first  **XL320** I put the id  `1 `and the second  **XL320**  I put the id  `2`, all dynamixels work well separately.
But I would like to control all Dynamixel at the same time and thats the problem went I put my code only 1 dynamixel move (XL320`ID1`) and the XL430 not react at all.
For the power supply I use the[ robotics starter kit](https://www.generationrobots.com/en/403718-official-dynamixel-starter-set-eu.html?utm_source=Doofinder&utm_medium=Doofinder&utm_campaign=Doofinder) power supply (12v 5a), but XL320 only uses 7.5v for that I put a voltage step-down [module ](https://www.amazon.fr/Yizhet-Convertisseur-Abaisseur-Alimentation-Regulateur/dp/B0823P6PW6/ref=pd_lpo_107_t_1/262-9509696-7430612?_encoding=UTF8&pd_rd_i=B0823P6PW6&pd_rd_r=4e7380a5-c492-4fa0-98be-7aab15a9aba1&pd_rd_w=OSPE3&pd_rd_wg=vw4Je&pf_rd_p=a9e8383d-b25d-45ec-acc2-a094dd781c31&pf_rd_r=CC1WHTT1NKAC5FRPV467&psc=1&refRID=CC1WHTT1NKAC5FRPV467)to convert 12v to 7.5v.
Thank you,
WilliamH07

WilliamH07, 2020-10-26:
Hello, 
I checked if all ID are the same in my code and my Dynamixel are the same, I checked if my baudrate and everything is exactly the same. Everything is ok, I don’t understand why my code isn’t working

ROBOTIS-Will, 2020-10-27:
Hi @WilliamH07 
Sorry about the late response.
Last but not least, could you tell us which controller do you use?
DYNAMIXEL2Arduino library and its example will work with OpenCR1.0 or OpenCM9.04.
If you are using DYNAMIXEL Shield with Arduino boards, you should use [DYNAMIXEL Shield library](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#dynamixel-shield-libraries).
Thank you.

WilliamH07, 2020-10-27:
Hello @ROBOTIS-Will thank for the reply,
I use a **[DynamixelShield](https://www.generationrobots.com/fr/403360-shield-dynamixel-pour-arduino.html?utm_source=Doofinder&utm_medium=Doofinder&utm_campaign=Doofinder)** with an Arduino Mega, I can control very Dynamixel independently with the library Dynamixel2Arduino.
I try the new library and I do like i already do in my old code. But like the first code, only one dynamixel works (XL320 `id1`) but not the (XL430 `id10`) 

My put my code below :
```
#include <DynamixelShield.h>

#if defined(ARDUINO_AVR_UNO) || defined(ARDUINO_AVR_MEGA2560)
  #include <SoftwareSerial.h>
  SoftwareSerial soft_serial(7, 8); // DYNAMIXELShield UART RX/TX
  #define DEBUG_SERIAL soft_serial
#elif defined(ARDUINO_SAM_DUE) || defined(ARDUINO_SAM_ZERO)
  #define DEBUG_SERIAL SerialUSB    
#else
  #define DEBUG_SERIAL Serial
#endif

const uint8_t DXL_ID2 = 10;
const uint8_t DXL_ID = 1;
const float DXL_PROTOCOL_VERSION = 2.0;

DynamixelShield dxl;

//This namespace is required to use Control table item names
using namespace ControlTableItem;

void setup() {
  // put your setup code here, to run once:
  
  // For Uno, Nano, Mini, and Mega, use UART port of DYNAMIXEL Shield to debug.
  DEBUG_SERIAL.begin(115200);

  // Set Port baudrate to 57600bps. This has to match with DYNAMIXEL baudrate.
  dxl.begin(1000000);
  // Set Port Protocol Version. This has to match with DYNAMIXEL protocol version.
  dxl.setPortProtocolVersion(DXL_PROTOCOL_VERSION);
  // Get DYNAMIXEL information
  dxl.ping(DXL_ID);
  dxl.ping(DXL_ID2);
  // Turn off torque when configuring items in EEPROM area
  dxl.torqueOff(DXL_ID);
  dxl.torqueOff(DXL_ID2);
  dxl.setOperatingMode(DXL_ID, OP_POSITION);
  dxl.setOperatingMode(DXL_ID2, OP_POSITION);
  dxl.torqueOn(DXL_ID);
  dxl.torqueOn(DXL_ID2);
}

void loop() {
  // put your main code here, to run repeatedly:
  
  // Please refer to e-Manual(http://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/) for available range of value. 
  // Set Goal Position in RAW value
  dxl.setGoalPosition(DXL_ID, 512);
  dxl.setGoalPosition(DXL_ID2, 512);
  delay(1000);
//  // Print present position in raw value
//  DEBUG_SERIAL.print("Present Position(raw) : ");
//  DEBUG_SERIAL.println(dxl.getPresentPosition(DXL_ID));
//  delay(1000);

  // Set Goal Position in DEGREE value
  dxl.setGoalPosition(DXL_ID, 5.7, UNIT_DEGREE);
  dxl.setGoalPosition(DXL_ID2, 5.7, UNIT_DEGREE);
  delay(1000);
  // Print present position in degree value
//  DEBUG_SERIAL.print("Present Position(degree) : ");
//  DEBUG_SERIAL.println(dxl.getPresentPosition(DXL_ID, UNIT_DEGREE));
  delay(1000);
}
```
Thank you,
WilliamH07

ROBOTIS-Will, 2020-10-28:
Hi @WilliamH07 
I tested your code and it worked well with my setting.
Please double check each DYNAMIXEL Configuration and `UART-DYNAMIXEL switch` on DYNAMIXEL Shield.
The switch should be on the `DYNAMIXEL` side when running the code.
Also, if you did not disconnect the VDD pin of XL-320, you might sharing the VDD for XL430 with XL-320 which will cause voltage error for XL-320.
[image]

WilliamH07, 2020-10-28:
Hello @ROBOTIS-Will , 
I use a step down voltage module to convert le 12v input to a 7.5v output for my XL320, I have 3 dynamixel already connected and ready to use. My XL430 are connected in chain, and my xl320 is on a connection direct to the board. I try to connect the XL430 without the chain, but only the XL320 works, I try the other one XL430 but only the XL320 works.
For the vdd I use the power DC jack input of the Arduino (12V), like I said before I use a step down voltage module to convert the 12v to a 7.5v for the xl320. (I'm working on a Niryo one Project)

You can see on this photo my dynamixel (XL430 in chain) and a XL320 on a direct connection to the board:

[image]

On this photo you can see my board and the hardware I use:

[image]
_(In the white box there is the step down module) with the cable from the XL320 and on the right there is the cable from the Xl430_

I use the dc jack port of the arduino to power on everything, I connected the pin 5V and GROUND of the connector (XL320) on the board to the step down module and the data pin I connected to the cable.

[image]

ROBOTIS-Will, 2020-10-29:
For me everything seems right.
Could you connect the U2D2 to one of the TTL port of DYNAMIXEL Shield and try scan?
During this scan, you should put UART switch to `Upload` side so that DYNAMIXEL wouldn't communicate with the Arduino board.
You may start scanning DYNAMIXEL with 2 XL430 and 2 XL-320.
If they are not detected, repeat the scan after removing one of them from the link.

WilliamH07, 2020-10-29:
Hello @ROBOTIS-Will,
I do the manipulation with the U2D2 like you said, every dynamixel works. DYNAMIXEL XL430 in link are recognized and the XL320 also. I'm actually working with one XL320, and two XL430. I use the DYNAMIXEL Wizard 2.0 for the software with the U2D2. I power up my dynamixel with the alimentation of the Arduino Mega. 

I post below my hardware:
[image]
Thank you,
WilliamH07

ROBOTIS-Will, 2020-11-02:
@WilliamH07 , if your Arduino doesn't do anything but supplying power, I don't see a reason for attaching DYNAMIXEL Shield to your Arduino board.
You can simply connect the SMPS to U2D2 PHB while leaving all the connection on the DYNAMIXEL Shield the same.
In this case, you should connect U2D2 and U2D2 PHB with a short 3 pin cable, and connect U2D2 PHB with DYNAMIXEL Shield with another 3 pin cable.

WilliamH07, 2020-11-02:
Hello @ROBOTIS-Will,
I try again to connect my **DynamixelShield** with the Arduino Board, to recognize the Dynamixel on the **U2D2**.
I power up my Dynamixel with the **Power Hub PCB U2D2**, very Dynamixel are recognized.
I can control very Dynamixel with the `Dynamixel Wizard 2.0` perfectly.

Like you said my code is working in your setting, you use a **OPENCM 9.04** with two **LBS-04** battery to do 7.5v for the **XL320** and for the **XL430** you use the DynamixelShield to power the **XL430** with 12v.

I use the 12v input from the **DynamixelShield**, then I convert it to 7.5v for the **XL320** and 12v for the **Xl430**, normally it's the exacly same configuration for me and you ? 

Thank you,
WilliamH07

ROBOTIS-Will, 2020-11-03:
I don't see anything wrong with your setting.
Please check if your DYNAMIXEL2Arduino and DYNAMIXEL Shield libraries are up to date and if possible, try supplying 7.5V from other source such as 2S Li-Po/Li-Io battery as I cannot assume that your voltage regulator works properly under all condition.

WilliamH07, 2020-11-04:
Hello @ROBOTIS-Will,
 I try to connect the power of the **XL320** to an other source of power, I connect the ground and data to the `DynamixelShield `and disconnect the **VDD** like you do.
 After that I connect the **XL430** in link and connect to the board, I put the 12V input for the **XL430**.
Everything power up perfectly, but only the `DXL_ID `works.

`const uint8_t DXL_ID = 1;
const uint8_t DXL_ID2 = 11;`

Because when I put the **ID1 or ID11** on the `DXL_1 `it's working, but when I put **ID1 or ID11** on the `DXL_ID2 `nothing works.

This is  my hardware (a little messy ) but its exacly what you do before in your post :

[image]
_Blue Green and Yellow cable are the ground. Orange cable is the Data. The blue cable on the left is for the 7.5v_
_Data and Ground are connected to the DynamixelShield._

The XL320 and the XL430 are on different power circuit.
Thank you,
WilliamH07

ROBOTIS-Will, 2020-11-05:
Sorry, but I still cannot replicate your issue.
Could you please confirm below information?

1. DYNAMIXEL2Arduino library version : 
2. DYNAMIXEL Shield library version :
3. Arduino IDE version :
4. XL-320 Firmware version :
5. XL-320 ID & Baudrate :
6. XL430 Firmware version :
7. XL430 ID & Baudrate :

WilliamH07, 2020-11-05:
Hello @ROBOTIS-Will,
I check everything you tell me to do :
1. DYNAMIXEL2Arduino library version is updated 
2. DYNAMIXEL Shield library version  is updated 
3. Arduino IDE version is updated
4. XL-320 Firmware version every XL320 are on the last version 
5. XL-320 ID & Baudrate `ID 001 1000000bps`
6. XL430 Firmware version every XL430 are on the last version 
7. XL430 ID & Baudrate `                                                                                                                                                                                                                                                 `

WilliamH07, 2020-11-10:
I try also to connect only the Xl430 and try to control them but it’s not working. I think it’s coming from the code dans not the hardware

ROBOTIS-Will, 2020-11-10:
The default baudrate of XL430 is 57600 so in order to use with XL-320, you should set it to 1000000bps.

WilliamH07, 2020-11-10:
The Xl430 and Xl320 are on the same baudrate (1000000bps)

WilliamH07, 2020-11-12:
And the ID are correctly set

WilliamH07, 2020-11-15:
Very ID are correctly set on the Dynamixel

ROBOTIS-Will, 2020-11-16:
If you only connected XL430 to the DYNAMIXEl Shield(without the XL-320 regulator) and commented out the unused XL-320 sections, but the code didn't work, your XL430 could be damaged.
It is not clear to identify your problem.
I'd recommend to uninstall and reinstall the libraries and reset all the hardware to factory setting, then try one at a time.

WilliamH07, 2020-11-18:
Hello @ROBOTIS-Will,
Thank you for the reply,
I test what you said to me, I factory reset the Dynamixel and  reinstall the library on an other PC to test if its not coming from my computer. The Dynamixel works one at a time, but went I try to move very Dynamixel at the same time, only one Dynamixel works. 
I'm pretty sure its coming for the code, because went I put the `XL430 ID10` on the `DXL_ID` its working,  but when I put the same Dynamixel on the `DXL_ID1` the same Dynamixel won't move. 
The Dynamixel are correctly set on the same baudrate for the communication and the ID match between the code and the reality.
Thank you,
WilliamH07

ROBOTIS-Will, 2020-11-19:
Hi @WilliamH07 
Could you also try with the baudrate of 57600 bps instead of 1Mbps? I haven't heard any issue with XL430 and XL-320, but it looks like some Arduino boards have slight margin of error for 1Mbps.
If you can also take the screenshot of transmitted / received packets with DYNAMIXEL Wizard 2.0 through U2D2, it would be helpful to figure out the issue.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#packet
1. Connect the U2D2 to any DYNAMIXEL 
2. Connect U2D2 to PC with a USB cable
3. Run DYNAMIXEL Wizard 2.0
4. Open the Packet window
5. Select the COM port and buadrate, then click `Open`
6. Run the Arduino example

I'm expecting to see something similar to below image.
[image]

WilliamH07, 2020-11-21:
Hello @ROBOTIS-Will ,
I try the code but I change the baudrate to `57600bps`, the **XL430** works at the same time.
But when I add to the code the **XL320**, only the **XL320** works, I change the **XL320** baudrate to `57600bps`.
The **XL430** who works at the beginning stop working with the **XL320**.
For the screenshot I don't understand, I connect the Dynamixel on the **U2D2** and I can read the packages but how I can run the arduino example when my motor are connected on the **U2D2** ? 
Thank you,
WilliamH07

ROBOTIS-Will, 2020-11-23:
In the previous comment, I used U2D2 to read the communication on DYNAMIXEL TTL line and used Packet feature of DYNAMIXEL Wizard 2.0.
You can follow the steps below to read the packet with U2D2.
1. Upload the sketch to Arduino Mega and connect XL430 and XL320 to DYNAMIXEL Shield.
2. Connect U2D2 to one of TTL ports on DYNAMIXEL Shield.
3. Connecto U2D2 to PC
4. Run DYNAMIXEL Wizard 2.0
5. Open `Packet` window and select the U2D2 port and DYNAMIXEL Baudrate(57600, in your most recent setup)
6. Check switches on DYNAMIXEL Shield. (DYNAMIXEL Power `ON`, UART sw to `Dynamixel` side)
7. Reset Arduino board.
8. Get the output on the Packet window.

WilliamH07, 2020-11-27:
Hello @ROBOTIS-Will,
Sorry for the late reply,
Thank you so much for your help all of this month, now everything is working perfectly fine XL320 and XL430 at the same time. 
If you want I can provide you the output of the packet.
Thank you so much for your help,
WilliamH07

ROBOTIS-Will, 2020-11-30:
Hi @WilliamH07 
I'm so glad to hear that your system works well :)
Could you share the cause of the issue for future customer support with similar case?
Thank you and wish the best for you.!

WilliamH07, 2020-12-01:
Hello @ROBOTIS-Will,
I don't now exactly what is the issue but I can provide a little tutorial to correct this problem :

**Do this step :** 

**1-** Connect the U2D2 on the Dynamixel shield with the Dynamixel connected
**2-** Plug in the power supply for the Dynamixel into the Arduino with the correct alimentation for the Dynamixel and connect the Arduino on the PC
**3-** Open the Dynamixel Wizard 2.0 and connect the U2D2 
**4-** Start the scan to recognize every Dynamixel
**5-** Make sure every Dynamixel are updated on the last version 
**6-** Make sure every Dynamixel are correctly set on the correct ID (Corresponding with the code)
**7-** Make sure every Dynamixel are set on the exactly same baudrate (put the Dynamixel on `57600bps`) 

Next set to make sure everything is okay,

- Open a new Arduino project, reinstall the `Dynamixel2Arduino library`, and paste your code on the new Arduino project.

- Make sure everything is corresponding between the **U2D2** and the Arduino Project (ID, Baudrate).

- Compile the code and upload it on the Arduino board, with the **Dynamixel Shield** switch set on **UPLOAD** and the **Dynamixel Shield** turn off. 

- Next when the code is uploaded on the Arduino board , on the **Dynamixel Shield** switch the switch on **Dynamixel** and turn on the **Dynamixel Shield**.

- And last, click on the button reset 

Normally everything is going to works fine but it's not working try the steps of @ROBOTIS-Will in the previous messages.

Thank you so much for your help,
WilliamH07

ROBOTIS-Will, 2020-12-02:
Thank you very much for the detailed step by step instruction.
We'll review your solution and see if we can find any bugs or issues in the code and trouble shooting process.
Appreciate your patience and hope you are safe from the Covid-19.
Thanks!

closed 2021-02-13 by WilliamH07