ROBOTIS-GIT/Dynamixel2Arduino #56: A problem of driving XM540-W270-R by using the youtube tutorial

AcarCopernicus, 2020-11-12:
I am following all the instruction in the tutorial, but the servo is only blinking. is there anyone can drive the servo XM540 by using dynamixel2arduino library? 

the video link;
https://www.youtube.com/watch?v=M7x7oWyOdGw&t=57s&ab_channel=ROBOTISCustomerSupport

ROBOTIS-Will, 2020-11-12:
Hi @AcarCopernicus ,
When using DYNAMIXEL Shield, please select the identical example from DYNAMIXEL Shield library instead of DYNAMIXEL2Arduino library.
Thank you.

AcarCopernicus, 2020-11-13:
Unfortunately, it did not solve my problem.

ROBOTIS-Will, 2020-11-16:
Could you elaborate more on your problem?

AcarCopernicus, 2020-11-16:
I have nothing to say in detail. I actually told every thing in the my first comment. I am trying to apply the same action in the video that is given below. But I can not drive servo by using arduino.

https://www.youtube.com/watch?v=M7x7oWyOdGw&t=57s&ab_channel=ROBOTISCustomerSupport

ROBOTIS-Will, 2020-11-17:
@AcarCopernicus 
Hi, you should at least give us some information about your hardware settings and details about the symptoms so that we can try replicating your issue. 
You could consider providing information such as
1. Which Arduino Board are you using?
2. What is the voltage of power supply?
3. Are you using a data transferable USB cable(some charge cable only supply power)?
4. Did you change any DYNAMIXEL settings?
5. Any error messages?
6. Are the libraries up to date?
7. Try uploading examples > DynamixelShield > basic > scan_dynamixel and see if you are getting any result.

AcarCopernicus, 2020-11-17:
Hello

Answers
1- I am using arduino uno r3
2-the power supply has 12V and 1.5A
3-I am using both USB cable and power supply cable at the same time
4-I do not  know how to change settings of Dynamixel. But I transferred data from arduino to dynamixel several times by changing baudrate. the servo motor just blinked
5-there is no errror messages
6-the newest libraries are uploaded from robotis git-hub
7- I tried, there is no response except from the blink.

ROBOTIS-Will, 2020-11-18:
@AcarCopernicus 
When uploading the sketch to Uno, you should set the `UART SW` to **Upload** side, and after the upload is completed, you should set the switch to **Dynamixel** side. This is because the Uno only has a single serial port, and this is why you should use [a separate USB serial interface](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#use-of-serial-monitor-with-dynamixel-shield) to read from the board.

What I meant for "7. Try scan_dynamixel example" was to see if you are getting any response from the terminal output. The example itself does not move the DYNAMIXEL.
DYNAMIXEL blinks once when powered so it is a normal behavior.

AcarCopernicus, 2020-11-18:
I tried as you described. there is nothing changed.

ROBOTIS-Will, 2020-11-19:
Could you write down each step you did so we can follow them and test?

AcarCopernicus, 2020-11-19:
Dear ROBOTIS-Will

I set the UART SW to Upload side  and I uploaded  the sketch to Uno, after the uploading operation finished, I set the UART SW to Dynamixel side.  I also set "ON" the shield. than the servo did nothing. so I checked the serial port screen. there is only "?" question mark. Could you please help me to solve this problem. I think this problem stem from a setting mistake. If you could help me by connnecting online using google meet or zoom, that is going to be a great help.

ROBOTIS-Will, 2020-11-20:
Hi,

If your code does not have any feature to hold the program until you are connected to the serial monitor, as soon as the upload is completed, it will start executing the code.
I'd recommend to try as below.
1. Turn on the Power Switch of DYNAMIXEL Shield. You don't need to turn this off when uploading the sketch.
2. Set the UART SW to Upload side.
3. Upload the sketch
4. Set the UART SW to Dynamixel side.
5. Reset the Arduino Uno board.

In the previous reply, I mentioned you need a separate USB serial interface
>  This is because the Uno only has a single serial port, and this is why you should use [a separate USB serial interface](https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#use-of-serial-monitor-with-dynamixel-shield) to read from the board.

You might also want to refer the below comment.
https://github.com/ROBOTIS-GIT/DynamixelShield/issues/18#issuecomment-730207992

If you are going to use Arduino Uno, you need additional USB-serial interface like [LN-101](https://emanual.robotis.com/docs/en/parts/interface/ln-101/). Or you could simply use Arduino Leonardo which comes with [a couple serial port](https://www.arduino.cc/reference/en/language/functions/communication/serial/).

I'm sorry but I'm not supposed to directly respond to customers special requests like online meeting, but you may request to schedule an online technical support via contactus2@robotis.com with more details about your project.
Thank you.