ROBOTIS-GIT/Dynamixel2Arduino #58: Doc says PROFILE_VELOCITY set is supported, it's not.

JamesNewton, 2021-02-09:
As a glance at:
https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/src/Dynamixel2Arduino.h#L438
shows, there is no way to set it, but the documentation at:
https://emanual.robotis.com/docs/en/parts/interface/dynamixel_shield/#library-api
shows that and a number of other parameters not supported by the library as being supported.

Please update docs or complete the todo list?

ROBOTIS-Will, 2021-02-10:
Hi @JamesNewton ,
The API does not contain any information about the [Profile Velocity](https://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#profile-velocity112), but does have for the [Goal Velocity](https://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#goal-velocity).
Although these two parameters look similar to each other, but they work quite differently.

In order to write the value in Profile Velocity address, you may use `writeControlTableItem()` as below.
```cpp
DynamixelShield dxl;
...
dxl.writeControlTableItem(PROFILE_VELOCITY, 1, 20);
```
Thank you.

JamesNewton, 2021-02-10:
No. 

I understand the difference between Profile Velocity and Goal Velocity. 

`dxl.writeControlTableItem(PROFILE_VELOCITY, 1, 20);` throws an error that PROFILE_VELOCITY is not defined for the XC-430 servos. Checking the code shows that PROFILE_VELOCITY is, in fact, NOT defined anywhere.

ROBOTIS-Will, 2021-02-10:
Have you included the namespace of `ControlTableItem` as I wrote in the above reply?
```cpp
using namespace ControlTableItem;
```

Without it, you'll get an error when trying to use the control table item index.
[image]

JamesNewton, 2021-02-10:
Ah! Perfect! This was exactly the issue. THANK YOU!

closed 2021-02-10 by JamesNewton