ROBOTIS-GIT/Dynamixel2Arduino #59: DynamixelShield SoftwareSerial DIRPIN

WilliamH07, 2021-02-13:
Hello guys 👍,
First of all, thanks for your amazing works it help me a lot in my progression. 
Actually, I use a Dynamixel Shield with a Cnc Shield on it for controlling Nema 17 stepper motor, and for controlling everything it's an Arduino Mega.
But, there is a problem, some of the motor use same pin of Dynamixel Shield in the code and it create conflit. The only way to solve this problem is to change the `RX TX` and `DIR PIN `of the Dynamixel Shield. 
In the library `SoftwareSerial `you can change the RX and TX, I try but it's not working, and for the Dynamixel Shield` DIRPIN` I don't know if I can change that. 
Can you tell me if it's possible to do ? And how I can to do it ? 
Thanks a lot,
WilliamH07

WilliamH07, 2021-02-14:
Hello guys,
Today I am trying something to fix this problem, I just solder the cables to the pins: 2;  7;  8 and I connect them to another pin of the mega Arduino.
And boom it works ! 
Thanks,
WilliamH07

closed 2021-02-14 by WilliamH07

ROBOTIS-Will, 2021-02-15:
Hi @WilliamH07 
FYI, Pin 2(DXL_DIR) is intended to control the data flow between Arduino board and DYNAMIXEL.
If DXL_DIR pin is not connected to DYNAMIXEL Shield, you may not properly read from or write to DYNAMIXEL.
Pin 7(SW_RX) and 8(SW_TX) does not require DIR control as it is a software emulated serial port.
Since it is an emulated serial port, higher baudrates are not supported.
Thank you.