ROBOTIS-GIT/Dynamixel2Arduino #61: slave.cpp processInstrWrite() no support for validating data other than ID and protocol version

sonelu, 2021-04-27:
When using `user_write_callback_` for custom defined registers we have the option to check the values and return a `packet_err` and the response packet produced by `processInstrWrite()` will correctly reflect this error. But the copy code [here](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/73c4d67819844c44622f04dc4dc00903c0d04556/src/utility/slave.cpp#L560) already updates the data in the register's values and these are not back-ed up. This means, although the response of the communication message is that there was an error, the internal register is updated with the fault data.

I think the backed up approach should be generalized and be performed before the copy of data and then in the restore part it could copy back the back-ed up data to the register in a generic way.

ROBOTIS-Will, 2021-04-28:
Hi @sonelu 
Thanks for the report.
I'm afraid that it may take long to update the code as the slave is not a major feature of this library, but I'll keep this thread open so that this issue can be resolved later.
You may create a PR for this if you'd like.
Thank you!

ROBOTIS-Will, 2021-08-11:
closing this thread as PR has been merged.

closed 2021-08-11 by ROBOTIS-Will