ROBOTIS-GIT/Dynamixel2Arduino #63: slave.cpp baudrate register

sonelu, 2021-04-29:
By definition a slave device should communicate over the Dynamixel bus, therefore I think it would be logical to include the baudrate as a default register in the class along with the ID, protocol, model number and firmware version.

I would recommend using the register id 8, which keeps it consistent with other devices. Also I would use the mapping table for baudrates based on the [latest devices](https://emanual.robotis.com/docs/en/dxl/p/ph54-200-s500-r/#baud-rate8). 

I think also we should implement the return delay time, as this is a common feature of the Dynamixel communication and all devices should accommodate it in case the master requires this delay. Typically the return delay time is register 9 that is already used by the protocol, so I suggest we use 10.

I can put a PR if this is ok.

ROBOTIS-Will, 2021-04-30:
Hi @sonelu 
Thanks for the suggestion.
I think adding the baudrate and return delay time are good practice as setting up a slave device.
I'll review and merge your PR.
Thanks!

jonghee-son, 2025-02-14:
Added in v0.4.8

closed 2025-02-14 by jonghee-son