ROBOTIS-GIT/Dynamixel2Arduino #64: setGoalVelocity with UNIT_PERCENT not mapping to actual max velocity

devdrik, 2021-05-07:
When using setGoalVelocity with unit set to UNIT_PERCENT I would expect it to be mapped to the servos max velocity. But instead the servo reaches max speed at around 21% and setGoalVelocity returns false for values above.

Tested on Arduino MKR WiFi 1010 with Dynamixel Shield for Arduino Series.
Servo: DYNAMIXEL XL330-M288-T
Power Supply: 5V (external through Shield)

ROBOTIS-Will, 2021-05-10:
Hi @devdrik 
Thank you for bring up this issue.
The reason that the velocity in percentage doesn't work correctly is that the library defines the maximum value based on the maximum available in the control table.
https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/src/Dynamixel2Arduino.cpp#L1128
However, the actual value is set to more realistic and effective value, therefore, entering to a higher percentage will be neglected from DYNAMIXEL.
I'll review this issue and see how the library can better explain the use of the function.
Thank you.

jonghee-son, 2025-03-19:
Hi @devdrik ,
Sorry for delayed response.
XL330-M288-T's maximum velocity limit is 2047 but initial velocity limit applied is 445, which is about 21% of maximum velocity limit.
To use setGoalVelocity with UNIT_PERCENT for speed faster than 21%, you have to change velocity limit.
Code block below is an example code to change velocity limit of XL330-M288-T.
```c
#include <Dynamixel2Arduino.h>

// Please modify it to suit your hardware.
/*******************************************************************************
* Copyright 2016 ROBOTIS CO., LTD.
*
* Licensed under the Apache License, Version 2.0 (the "License");
* you may not use this file except in compliance with the License.
* You may obtain a copy of the License at
*
*     http://www.apache.org/licenses/LICENSE-2.0
*
* Unless required by applicable law or agreed to in writing, software
* distributed under the License is distributed on an "AS IS" BASIS,
* WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
* See the License for the specific language governing permissions and
* limitations under the License.
*******************************************************************************/

#include <Dynamixel2Arduino.h>

// Please modify it to suit your hardware.
#if defined(ARDUINO_AVR_UNO) || defined(ARDUINO_AVR_MEGA2560) // When using DynamixelShield
  #include <SoftwareSerial.h>
  SoftwareSerial soft_serial(7, 8); // DYNAMIXELShield UART RX/TX
  #define DXL_SERIAL   Serial
  #define DEBUG_SERIAL soft_serial
  const int DXL_DIR_PIN = 2; // DYNAMIXEL Shield DIR PIN
#elif defined(ARDUINO_SAM_DUE) // When using DynamixelShield
  #define DXL_SERIAL   Serial
  #define DEBUG_SERIAL SerialUSB
  const int DXL_DIR_PIN = 2; // DYNAMIXEL Shield DIR PIN
#elif defined(ARDUINO_SAM_ZERO) // When using DynamixelShield
  #define DXL_SERIAL   Serial1
  #define DEBUG_SERIAL SerialUSB
  const int DXL_DIR_PIN = 2; // DYNAMIXEL Shield DIR PIN
#elif defined(ARDUINO_OpenCM904) // When using official ROBOTIS board with DXL circuit.
  #define DXL_SERIAL   Serial3 //OpenCM9.04 EXP Board's DXL port Serial. (Serial1 for the DXL port on the OpenCM 9.04 board)
  #define DEBUG_SERIAL Serial
  const int DXL_DIR_PIN = 22; //OpenCM9.04 EXP Board's DIR PIN. (28 for the DXL port on the OpenCM 9.04 board)
#elif defined(ARDUINO_OpenCR) // When using official ROBOTIS board with DXL circuit.
  // For OpenCR, there is a DXL Power Enable pin, so you must initialize and control it.
  // Reference link : https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/DynamixelSDK/src/dynamixel_sdk/port_handler_arduino.cpp#L78
  #define DXL_SERIAL   Serial3
  #define DEBUG_SERIAL Serial
  const int DXL_DIR_PIN = 84; // OpenCR Board's DIR PIN.    
#elif defined(ARDUINO_OpenRB)  // When using OpenRB-150
  //OpenRB does not require the DIR control pin.
  #define DXL_SERIAL Serial1
  #define DEBUG_SERIAL Serial
  const int DXL_DIR_PIN = -1;
#else // Other boards when using DynamixelShield
  #define DXL_SERIAL   Serial1
  #define DEBUG_SERIAL Serial
  const int DXL_DIR_PIN = 2; // DYNAMIXEL Shield DIR PIN
#endif
 

const uint8_t DXL_ID = 1;
const float DXL_PROTOCOL_VERSION = 2.0;

Dynamixel2Arduino dxl(DXL_SERIAL, DXL_DIR_PIN);

//This namespace is required to use Control table item names
using namespace ControlTableItem;

void setup() {
  // put your setup code here, to run once:
  
  // Use UART port of DYNAMIXEL Shield to debug.
  DEBUG_SERIAL.begin(115200);
  while(!DEBUG_SERIAL);
  
  // Set Port baudrate to 57600bps. This has to match with DYNAMIXEL baudrate.
  dxl.begin(57600);
  // Set Port Protocol Version. This has to match with DYNAMIXEL protocol version.
  dxl.setPortProtocolVersion(DXL_PROTOCOL_VERSION);

  DEBUG_SERIAL.print("PROTOCOL ");
  DEBUG_SERIAL.print(DXL_PROTOCOL_VERSION, 1);
  DEBUG_SERIAL.print(", ID ");
  DEBUG_SERIAL.print(DXL_ID);
  DEBUG_SERIAL.print(": ");
  if(dxl.ping(DXL_ID) == true) {
    DEBUG_SERIAL.println("ping succeeded!");
    
    // Turn off torque when configuring items in EEPROM area
    dxl.torqueOff(DXL_ID);
    
    // Set velocity limit to maximum value (2047 for XL330)
    dxl.writeControlTableItem(VELOCITY_LIMIT, DXL_ID, 2047);
    DEBUG_SERIAL.print("Current velocity limit: ");
    DEBUG_SERIAL.println(dxl.readControlTableItem(VELOCITY_LIMIT, DXL_ID));
  }
  else{
    DEBUG_SERIAL.println("ping failed!");
  }
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
Thank you.

jonghee-son, 2025-03-28:
Closing this issue due to no recent activities.
Feel free to reopen this issue if you need more assistance.

closed 2025-03-28 by jonghee-son