ROBOTIS-GIT/Dynamixel2Arduino #73: OPERATING_MODE enums not correct numbers?

LegacyMecha, 2021-07-27:
When I use "mode = dynamixel.readControlTableItem(OPERATING_MODE, id)" on an XH430, I get "3" when it's set to Position Mode, but in code, when I check "mode == OP_POSITION" OP_POSITION is 0 and OP_VELOCITY is actually "3". Am I doing something wrong? On Serial.print(OP_POSITION) = 0, but in DynamixelWizard "Operating Mode(11)" = 3 Position Mode.

ROBOTIS-Will, 2021-07-28:
Hi @LegacyMecha 

You are correct, and there's nothing wrong in your code.
As you mentioned, the [actual value for the Position Control Mode is 3](https://emanual.robotis.com/docs/en/dxl/x/xh430-w350/#operating-mode11).
In the DYNAMIXEL2Arduino library, the operating mode is defined by a enumeration which begins with OP_POSITION = 0 as below.
https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/master/src/Dynamixel2Arduino.h#L25

Since this could be confusing, I'll list up this in the to do list.
Thank you!

LegacyMecha, 2021-07-28:
It would be nice to have a function maybe that can check which mode a servo is in. Like getOperatingMode(id) and then do a compare (==) to OP_VELOCITY, OP_POSITION...etc.

ROBOTIS-Will, 2021-07-29:
@LegacyMecha 
Thanks for your feedback.
I'll keep updating the status via this thread.

commit 61864bd1 (2021-08-12): add new example, fix #73
closed 2021-08-12 by ROBOTIS-Will