ROBOTIS-GIT/Dynamixel2Arduino #75: Dynamixel MX-28T and MX-106T (2.0)+Arduino Uno+IDE_Extended Position Control Mode+Velocity limit

Danaish-Rathore, 2021-08-06:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.4.1`

2. Which programming language/tool do you use?
   Arduino

3. Which operating system do you use?
   Windows 10

3. Which USB serial converter do you use?
  Arduino Uno

4. Which DYNAMIXEL do you use?
   MX-28T(2.0) 

5. Have you searched the issue from the closed issue threads?

6. Please describe the issue in detail

Hi everyone, 

I am  controlling **MX servos** using Arduino Uno and working on the **Extended position control mode**, now the problem is I don't know how to  set the **velocity limit**/control velocity for MX servo _protocol (2.0)_ under extended position control mode. However, I have tried with the below code and it seems the code is working and the velocity is controlled only for _protocol (1.0)_ .   

`void setup() {
  
  DEBUG_SERIAL.begin(115200);
  dxl.begin(57600);
  dxl.setPortProtocolVersion(DXL_PROTOCOL_VERSION);
  dxl.ping(DXL_ID);
  
  // Turn off torque when configuring items in EEPROM area
  dxl.torqueOff(DXL_ID);
  dxl.setOperatingMode(DXL_ID, OP_VELOCITY);
  dxl.torqueOn(DXL_ID);
  
  //Set Goal Velocity using RPM
  dxl.setGoalVelocity(DXL_ID, 20, UNIT_RPM);
  delay(100);

  DEBUG_SERIAL.print("Present Velocity(rpm) : ");
  DEBUG_SERIAL.println(dxl.getPresentVelocity(DXL_ID, UNIT_RPM));
  delay(1000);
  
  
}

void loop() {
 
  
  // Extended Position Control Mode in protocol2.0, 
  //dxl.torqueOff(DXL_ID);
  dxl.setOperatingMode(DXL_ID, OP_EXTENDED_POSITION);
  dxl.torqueOn(DXL_ID);
  if(dxl.setGoalPosition(DXL_ID, 0)){
    delay(5000);
    DEBUG_SERIAL.print("Present Extended Position1 : ");
    DEBUG_SERIAL.println(dxl.getPresentPosition(DXL_ID)); DEBUG_SERIAL.println();
  }
  // Extended Position Contorl Mode in protocol2.0, 
   //dxl.torqueOff(DXL_ID);
  dxl.setOperatingMode(DXL_ID, OP_EXTENDED_POSITION);
  dxl.torqueOn(DXL_ID);
  if(dxl.setGoalPosition(DXL_ID, 5000)){
    delay(5000);
    DEBUG_SERIAL.print("Present Extended Position2 : ");
    DEBUG_SERIAL.println(dxl.getPresentPosition(DXL_ID)); DEBUG_SERIAL.println();
  }
}`

The code can control the extended position and velocity of servo having version (0.1), but the same code failed for version (2.0) because the dynamixel servo (2.0) rotates with max.velocity (where as it's velocity should be according to the goal velocity) following the extended position limits. 

Please, if someone knows what wrong I am doing! and how to modify the code... it will be great help and thanks.

Regards,
Danaish Rathore

7. How can we reproduce the issue?

ROBOTIS-Will, 2021-08-06:
@Danaish-Rathore 
I've transferred this issue to the relevant repository.
Thanks!

Danaish-Rathore, 2021-08-06:
**@ROBOTIS-Will** 

Again many thanks. Actually I tried the profile velocity yesterday night and it worked but I couldn't update here.

ROBOTIS-Will, 2021-08-09:
@Danaish-Rathore 
Oh, I see. Glad that the Profile Velocity works for you.
Not sure why the transferred thread cannot be updated by the original author, but thanks for the updates! :)

Danaish-Rathore, 2021-08-09:
**@ROBOTIS-Will** 
The profile velocity only works for protocol (2.0) and it doesn't work for MX-28T protocol (1.0). Therefore, for version (1.0) the goal velocity can be used with extended position control mode as I tried. Thanks

ROBOTIS-Will, 2021-08-10:
@Danaish-Rathore 
The MX-28T with Protocol 1.0 does not have Profile Velocity / Acceleration address.
In case of using Protocol 1.0 for MX series, you can control the max velocity with Moving Speed(32)
```cpp
dxl.writeControlTableItem(PROFILE_VELOCITY, DXL_ID, 100);
```

Danaish-Rathore, 2021-08-11:
@ROBOTIS-Will 

Oh! that is nice. Next time I will try with this and again many thanks.

closed 2021-09-04 by Danaish-Rathore