ROBOTIS-GIT/Dynamixel2Arduino #89: setOperatingMode does not work

THEb0nny, 2021-12-10:
I set the hinge mode to the 3rd ax-12a in the loop, but as it turned out that motors with id 2 and 3 did not set this mode.
All motors were connected and the port monitor displayed the model number according to their ID.
The motors were not working correctly, the motor was working correctly 1.
Then I opened the RoboPlus application and looked at their mode of operation. 2 and 3 had wheel mode.

ROBOTIS-Will, 2021-12-13:
Hi @THEb0nny 

Could you give provide more information about the items below?
- Which controller are you using?
- What is your power source spec?
- Have you turned off the Torque before setting the operating mode? (You cannot write on EEPROM while Torque is turned on)
- If possible, please copy and paste snippet of your code.

THEb0nny, 2021-12-13:
@ROBOTIS-Will I am using opencm 9.04-c with OpenCM 485 EXP.
The power supply is a Robiton power supply with various power options. I've tried 9 and 12 volts.
I don't understand what point 3 means.
`
#include <Dynamixel2Arduino.h>  // Подключение библиотеки Dynamixel

#define DEBUG_SERIAL Serial // Установка константы, отвечающей за последовательный порт, подключаемый к компьютеру
#define DXL_SERIAL Serial3 // OpenCM9.04 EXP Board's DXL port Serial. (To use the DXL port on the OpenCM 9.04 board, you must use Serial1 for Serial. And because of the OpenCM 9.04 driver code, you must call Serial1.setDxlMode(true); before dxl.begin();.)

#define DXL_DIR_PIN 22 // Инициализация переменной, отвечащей за номер пина, подключенного к информационному пину приводов манипулятора
#define DXL_PROTOCOL_VERSION 1.0 // Инициализация переменной, отвечащей за протокол передачи данных от OpenCM9.04 к приводам

#define JOINT_N 3 // Количество приводов

Dynamixel2Arduino dxl(DXL_SERIAL, DXL_DIR_PIN); // Инициализация указателя на команды из библиотеки Dynamixel

void setup() {
  Serial.begin(57600); // Установка скорости обмена данными по последовательному порту компьютера
  while(!DEBUG_SERIAL); // Ждём, пока монитор порта не откроется
  Serial.println("Setup...");
  dxl.begin(1000000); // Установка скорости обмена данными по последовательному порту манипулятора
  dxl.setPortProtocolVersion(DXL_PROTOCOL_VERSION); // Выбор протокола обмена данными
  delay(2000); // Ждём, чтобы моторы успели подключиться
  for (int i = 1; i <= JOINT_N; i++) { // Цикл для перебора всех приводов
    dxl.setOperatingMode(i, OP_POSITION); // Установка режима работы привода в качестве шарнира
    if(dxl.ping(i) == true) { // Проверка отвечает ли мотор
      Serial.print("Dynamixel with ID "); Serial.print(i); Serial.print(" found, model "); Serial.print(dxl.getModelNumber(i)); Serial.println(".");
    } else {
      Serial.print("Dynamixel with ID "); Serial.print(i); Serial.println(" not found!");
    }
  }
  delay(500);
}

void loop() {
  for (int i = 1; i <= JOINT_N; i++) {
    dxl.setGoalVelocity(i, 50); // Задание целевой скорости 50 приводу с номером i
    dxl.setGoalPosition(i, 410); // Задание целевого положения -30 градусов приводу с номером i
  }
  delay(2000); // Задержка, пока приводы займут нужное положение

  for (int i = 1; i <= JOINT_N; i++) {
    dxl.setGoalVelocity(i, 50); // Задание целевой скорости 50 приводу с номером i
    dxl.setGoalPosition(i, 614); // Задание целевого положения +30 градусов приводу с номером i
  }
  delay(2000); // Задержка, пока приводы займут нужное положение
}
`

THEb0nny, 2021-12-15:
> @THEb0nny 
> Thanks for the code.
> 
> Is there a reason of commenting out several definitions in the beginning of the code?
> There are some critical definitions, so you should leave them uncommented when using DYNAMIXEL2Arduino library.
> 
> Not sure what went wrong other than that.
> Do you properly get the ping from each DYNAMIXEL?
> Could you check if applying a delay(10) after the setOperatingMode() helps?

I always write comments in define and I never had a problem. Can be removed.

The ping from the speakers works well. The motors are now tuned with cm-530. Each motor responds with its own ID and writes the model number.

ROBOTIS-Will, 2021-12-19:
Oops, my bad.

My C++ intellisense wasn't working and the definitions just had the same text color with the comments so I thought they were commented out.

> Each motor responds with its own ID and writes the model number.

So everything works fine now?

THEb0nny, 2022-05-16:
> Oops, my bad.
> 
> My C++ intellisense wasn't working and the definitions just had the same text color with the comments so I thought they were commented out.
> 
> > Each motor responds with its own ID and writes the model number.
> 
> So everything works fine now?

Yes, everything works for us. Sorry I forgot to reply...
=
It was necessary to turn off the torque so that memory writing could be carried out.

closed 2025-02-14 by jonghee-son