ROBOTIS-GIT/Dynamixel2Arduino #93: Seeeduino XIAO RP2040 outputs wrong CRC in packet

Deezmaker, 2022-02-24:
I wrote a simple arduino sketch to ping ID#1 for the Seeeduino XIAO and it works great. When I compile the exact same sketch for the XIAO RP2040 the output is different on the CRC value in packet as follows:

Seeeduino XIAO = "FF FF FD 0 1 3 0 1 19 4E" (which is correct and the dynamixel responds to)

Seeeduino XIAO RP2040 = "FF FF FD 0 1 3 0 1 2D 1" (wrong CRC?)

Can you guys look into this?

Deezmaker, 2022-05-18:
Looks like Seeeduino folks fixed this in latest board library version. Not an issue anymore.

ROBOTIS-Will, 2022-05-19:
@Deezmaker 
Missing the last couple bytes is usually due to the prematured change of the TX/RX direction.
Glad that this has been fixed.

closed 2025-02-14 by jonghee-son