ROBOTIS-GIT/Dynamixel2Arduino #96: Communication Problem between Portenta H7 and MKR Dynamixel Shield

Matt93-00, 2022-03-22:
I noticed some time ago that even if all codes, coming from Dynamixel2Arduino or DynamixelShield, compile well, in the end there is no way to find the dynamixel attached to the system Portentah7+MKRDynamixelShield. Thus, I can't drive the dynamixel with portenta H7. 
Thank You for the availability.

ROBOTIS-Shibata, 2022-04-04:
It's not my experience, but I know of other cases the Portenta H7 + DYNAMIXEL Shield for Arduino MKR series can't be worked.

ROBOTIS-Will, 2022-04-29:
I'm sorry about the delayed response on this issue.
A temporary fix for this issue until the Arduino mbed library is updated can be found in the below commit.
https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/commit/2800ae9dacb6949ddacff65131ab100a1af0b26b
Thank you.

Matt93-00, 2022-05-02:
Hi @ROBOTIS-Will , thank You so much for your availability. Anyway, I shall say that your solution doesn't work for me. I changed the .cpp and .h files with your edited files and nothing happens again. 0 dynamixel are found during scanning and no one code works! Did I complete the right steps? 
@shiba-8ro Did you test it?

ROBOTIS-Will, 2022-05-03:
@Matt93-00 
Could you specify your development environment and which product do you use?
Have you correctly located the DYNAMIXEL2Arduino library directory?
If you turned off the `Use external editor` option in Arduino IDE preference, you should restart the Arduino IDE after modifying the library.

Matt93-00, 2022-05-03:
@ROBOTIS-Will  Yes, sure! I hope that this can help other people.
I am using Arduino IDE on Windows 10. The "use external editor" is always not flagged. To be complete, I have a Portenta h7 (https://store.arduino.cc/products/portenta-h7?_gl=1*1hncf6w*_ga*OTcyNjgzMjk3LjE2MTM3MzI0MzI.*_ga_NEXN8H46L5*MTY1MTU2MjE4Ni40NDUuMS4xNjUxNTYyMjQ5LjA.) attached to the MKR shield with MX64AT dynamixel, connected as documentation said at paragraph 5.2 https://emanual.robotis.com/docs/en/parts/interface/mkr_shield/

jonghee-son, 2025-04-01:
Hi @Matt93-00 

Did PR #103 fixed your problem?
If not, feel free to reach out to us for more assistance.

Thank you.

PR #103 (2022-05-10): add tx_en delay

fixes #96
closed 2025-04-26 by jonghee-son