ROBOTIS-GIT/Dynamixel2Arduino #108: Using SPI_1 and SPI_2 on OpenCM-904 in Arduino mode

Roboteer1000, 2022-07-16:
In Arduino mode, the OpenCM-904 already has drawn out nicely 2 SPI Ports on its two headers.
In the file "SPI.h", Line 114, I saw that SPI is defined as **extern SPIClass SPI;**
Then in the file "SPI.cpp", Line 25, I saw that only SPI1 is used by default.  So I added SPI2 on Line 26 (and commented out Line 25).

[image]

With this setup, I can use **EITHER SPI1 or SPI2 Port fine**, but I need to use BOTH SPI Ports at the same time for my project.  Header pins wise, there are no problems, but software wise how should I approach this issue?  

I also saw the comments on Lines 63-66 - is that relevant/workable?

[image]

The way multiple SPI devices are handled as shown in "Arduino Cookbook" is very different from the way ROBOTIS is handling this issue for OpenCM-904.  The use of "extern SPIClass SPI" - is that really needed?  Did ROBOTIS write the OpenCM-904 library so that only ONE SPI can be used at any one time? 

Thank you for any pointer.

jonghee-son, 2025-02-27:
Hi @Roboteer1000,
If you want to use 2 separate SPI for your projects, you can initialize 2 SPI class like this code below.
```c
SPIClass my_spi1(SPI1);
SPIClass my_spi2(SPI2);
```
And both SPI1 and SPI2 worked fine during our test.
[image]
If there's any other problem, please let us know.
Thank you.

Roboteer1000, 2025-03-02:
I see, just needed a separate handle for each SPI port.  Thank you so very much for the help.

closed 2025-03-02 by Roboteer1000