ROBOTIS-GIT/Dynamixel2Arduino #127: UNO R4 - Library(libraries) - will not build on new Arduino boards... And may not work at all.

KurtE, 2023-07-24:
Recently I started playing with the new Arduino UNO R4 boards (MINIMA and WIFI) - and yesterday I thought I would try it 
So I pulled out my Dynamixel Servo Shield and plugged it into the Minima.  I made sure my libraries were up to date.

Loaded the scan_dynamixel.ino example sketch and tried to compile it which failed.  More details in the forum thread:
https://forum.arduino.cc/t/robotis-dynamixel-shield-code-does-not-compile/1150975

Found the issue was that Arduino defined: LED_RED and LED_BLUE
```
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\variants\MINIMA\includes\ra_cfg\fsp_cfg\bsp\bsp_pin_cfg.h:
   10  #define A1 (BSP_IO_PORT_00_PIN_01)
   11  #define A2 (BSP_IO_PORT_00_PIN_02)
   12: #define LED_RED (BSP_IO_PORT_00_PIN_11)
   13  #define TX_LED (BSP_IO_PORT_00_PIN_12)
   14  #define LED_BLUE (BSP_IO_PORT_00_PIN_13)

C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\variants\UNOWIFIR4\includes\ra_cfg\fsp_cfg\bsp\bsp_pin_cfg.h:
   10  #define A1 (BSP_IO_PORT_00_PIN_01)
   11  #define A2 (BSP_IO_PORT_00_PIN_02)
   12: #define LED_RED (BSP_IO_PORT_00_PIN_11)
   13  #define TX_LED (BSP_IO_PORT_00_PIN_12)
   14  #define LED_BLUE (BSP_IO_PORT_00_PIN_13)

2 matches across 2 files
```
Which completely screws up anyone else who use these words in any way, including yours which should be safe,
as yours are contained in an enum, which is in a namespace, which is nice and clean.

I created an issue yesterday: https://github.com/arduino/ArduinoCore-renesas/issues/64
Which they blew off and closed it with the response:
`I suggest you open an issue with the Dynamixel library, we are likely going to keep those defines.`

So here it is.

Not sure how many libraries, documents, examples, and the like this might impact.  Nor if there are any other now new magic words, which may show up with additional testing.

Note: currently even if you "fix" this, their serial subsystem has lots of issues, including Serial1.flush() does not work, and as such the direction pin will change state before it should, and the packets are corrupted. 
For example I had 18 servos and it only found 5... Some of this could be wiring connection issues, but mostly issues with their subsystems. 

I have experimented with some updates to their serial code base (https://github.com/arduino/ArduinoCore-renesas/pull/59), but this has not gained any traction, so not expecting that any of these issues will be fixed anytime soon.

ROBOTIS-Will, 2023-07-25:
@KurtE 

Hi, it's been a while talking to you :)
Yes, there are some definitions that are identical from the Arduino boards, and LED colors.
It seems even if we make modifications on those definition, it may not affect to our library.
However, there were a few cases when the Arduino users wanted to keep the LED_COLOR function so that they can control their own LED lights for other purpose.

I'll see if I can find some other workaround to resolve this issue.
Thanks for reporting

KurtE, 2023-07-26:
@ROBOTIS-Will - Yes been awhile.  Hope all is well!

Been mostly playing around with Teensy boards and the like.  Thought I would try out the new UNO boards.
They feel far more like an Alpha or Beta release than a real production release.

As for preserving the usage of their #defines, I personally would not worry much about that. 
They maybe have a slight possibility of usage on MINIMA (keyword slight)

The defines are the same for both boards:
```
#define LED_RED (BSP_IO_PORT_00_PIN_11)
#define TX_LED (BSP_IO_PORT_00_PIN_12)
#define LED_BLUE (BSP_IO_PORT_00_PIN_13)
```
On MINIMA - P011 is not connected, P012 is connected to TX , P013 is connected to RX .

On WIFI - P011-P013 are connected to their 8x12 led matrix, which require knowledge on the setting of the different IO pins on the matrix.  Charlieplexing 

Locally I have mine hacked up like:
```
// -----------------------------------------------
// Kurts hack
#ifdef LED_RED
#undef LED_RED
#undef LED_BLUE
#endif

namespace ControlTableItem{
  enum ControlTableItemIndex{

```
Again hope all is well.

turiphro, 2024-08-02:
Hey! This issue still seems to be around. I get the same compile errors for the Arduino UNO R4 WiFi board. Adding the `undef` hack to actuator.h seems to fix the issue.

Happy to submit a pull request to add the `undef` statements if that's all what is needed.

closed 2025-02-14 by sookyungson