ROBOTIS-GIT/Dynamixel2Arduino #132: Converting to Units when data is read with syncRead and BulkRead

SupSonAlv, 2023-09-18:
First of all, hi:
I don't know if it's right to ask this here. If it is not the right location, I would appreciate it if you could give me an idea for the relevant location.

-The servo motor I use is mx-106 (protocol 2.0).
-I am using the OpenCR 1.0 card.

-In the library
getPresentPosition
getPresentVelocity
getPresentCurrent
With its functions, it can directly convert the data into the required units and export it.
But I want to do this for data read with SyncRead.
-From two servos with SyncRead
present_current;
present_velocity;
present_position;

I can read data from addresses with syncRead(I think it can also be used for BulkRead) and I want to convert this data into units. How can I do this? How can I use a built-in function for this?

Every opinion given is worth respecting.

dhood, 2025-02-17:
I'm also interested in this for our project, sync read is just giving raw units I think?

jonghee-son, 2025-02-24:
Hi @SupSonAlv ,
To format raw data from SyncRead, you can use method from checkAndconvertReadData function for simple read actions.
In Dynamixel2Arduino.cpp, checkAndconvertReadData is declared like below.
```c
static bool checkAndconvertReadData(int32_t in_data, float &out_data, uint8_t unit, ItemAndRangeInfo_t &item_info)
{
  float data = 0;  

  switch(unit)
  {
  case UNIT_RAW:
    data = (float)in_data;
    break;

  case UNIT_PERCENT:
    data = (float)f_map(in_data, 
     (float)item_info.min_value, (float)item_info.max_value, -100.0, 100.0);
    break;    

  case UNIT_RPM:
  case UNIT_DEGREE:
  case UNIT_MILLI_AMPERE:
    if(unit != item_info.unit_type)
      return false;

    data = (float)in_data*item_info.unit_value;
    break;

  default:
    return false;
  }

  out_data = data;

  return true;
}
```
To modify this code to fit your need, you have to change values related Dynamixel specific item_info.
For example, if you want to convert raw position data into unit percent, you have to look up for minimum and maximum value of position data for your Dynamixel.
In case of MX-106T/R model, the minimum and maximum value for position data is 0 and 4096 for each.
(You can find this in our [E-Manual](https://emanual.robotis.com/docs/en/dxl/mx/mx-106/#control-table-of-eeprom-area))
To use this values you can write a code like below.
```c
float f_map(float x, float in_min, float in_max, float out_min, float out_max) {
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}

int32_t formated_data = (float)f_map(input_raw_data, (float)0, (float)4096, -100.0, 100.0);
```
Thank you.